wasm-micro-runtime/wasm-micro-runtime #263: Retrieving returned buffers from WASM app

damgaard22, 2020-05-24:
I read the documentation section on passing buffers to a wasm app, though I saw no mention of retrieving a returned buffer from a wasm app. 

My wasm app returns a Uint8Array.

Is this possible in WAMR?

Rochet2, 2020-05-24:
Do you mean a function similar to this?
```cpp
uint8_t my_array[5] = {0};
uint8_t* GetArray() {
    return my_array;
}
```

To return an array you would need to return the array's location (the pointer) and possibly the length of it. Assuming you already know that the length is 5 (possibly through a GetLength function call) you can do the following in WAMR:

```cpp
  func = wasm_runtime_lookup_function(module_inst, "GetArray", NULL);
  exec_env = wasm_runtime_create_exec_env(module_inst, stack_size);
  uint32_targv[2]; // I assume this always has to be at least 2 to account for possible 64 bit return value
  iwasm_runtime_call_wasm(exec_env, func, 0, argv);
  /* the return value is stored in argv[0] */
  uint32_t index = argv[0]; // pointer is equal to memory position
  uint32_t length = 5;
  if (!wasm_runtime_validate_app_addr(index, length))
    return; // safety check that we dont read out of bounds memory
  array = (uint8_t*)wasm_runtime_addr_app_to_native(index);
  // you can now use array and read/write to it, but make sure you keep within the length of 5.
```
In this code we call the GetArray function. We then take the returned pointer value which is an index to WASM memory, We use the index and length to validate that it is indeed valid memory and finally convert the memory index to a real pointer. Using the pointer and the length we can finally read and write to the array that exists in WASM memory. To simplify the code, error checking code was left out from the function call.

This is based on
https://github.com/bytecodealliance/wasm-micro-runtime/blob/ef5ceffe71ee3a9429e72b76f1e4d94c26d90249/doc/embed_wamr.md#native-calls-wasm-functions-and-passes-parameters
and
https://github.com/bytecodealliance/wasm-micro-runtime/blob/ef5ceffe71ee3a9429e72b76f1e4d94c26d90249/doc/export_native_api.md#buffer-address-conversion-and-boundary-check

mdamgaard-biolib, 2020-05-24:
This looks really promising.

Do you know of any way to return the pointer and the length of the array as you suggest in the first paragraph?

damgaard22, 2020-05-24:
This looks really promising!

Do you know of a way to return both the pointer and the length of the array as you suggest in the beginning?

Rochet2, 2020-05-24:
WASM can currently work with only one return value.
The length and pointer would need to be fetched by separate function calls or by encoding both of them into the single value returned by the function. Here are a few possibilities to return length and location in one function call:

WASM supports i64 currently, allowing you to  return for example uint64 value that contains both the length and  the pointer value, as both could be uint32.
```cpp
uint8_t my_array[5] = {0};
uint64_t GetArray() {
    uint32_t ptr = my_array;
    uint32_t length = sizeof(my_array);
    uint64_t coded = ptr;
    coded << 32;
    return coded + ptr;
}
```
```cpp
  func = wasm_runtime_lookup_function(module_inst, "GetArray", NULL);
  uint32_t argv[2] = {0};
  iwasm_runtime_call_wasm(exec_env, func, 0, argv);
  uint32_t index = argv[0];
  uint32_t length = argv[1];
  if (!wasm_runtime_validate_app_addr(index, length))
    return;
  array = (uint8_t*)wasm_runtime_addr_app_to_native(index);
  // you can now use array and length to read/write to it.
```

Alternatively you can include the length to the buffer, possibly for example with the following:
```cpp
struct Data {
  uint32_t length = 5;
  uint8_t my_array[5] = {0};
}
Data data;

Data* GetArray() {
    return &data;
}
```
```cpp
  func = wasm_runtime_lookup_function(module_inst, "GetArray", NULL);
  uint32_t argv[2];
  iwasm_runtime_call_wasm(exec_env, func, 0, argv);
  /* the return value is stored in argv[0] */
  uint32_t index = argv[0]; // pointer is equal to memory position
  uint32_t length;

  // Get length
  uint32_t array_length_length = 2;
  if (!wasm_runtime_validate_app_addr(index, array_length_length))
    return;
  array_length = *((uint32_t*)wasm_runtime_addr_app_to_native(index));

  // Get array
  if (!wasm_runtime_validate_app_addr(index+array_length_length, array_length))
    return;
  array = (uint8_t*)wasm_runtime_addr_app_to_native(index+array_length_length));
  // you can now use array and array_length to read/write to it.
```

Both of these approaches could have flaws that I am unaware of. There are a lot of assumptions at play here with how the data is represented in memory at runtime.

mdamgaard-biolib, 2020-05-24:
This seems to be consistent with how i thought it would work.
Thank you very much, i will give it a whirl and close the issue when i get it working.

damgaard22, 2020-05-24:
This seems to be consistent with how i thought it would work.
Thank you very much, i will give it a whirl and close the issue when i get it working.

closed 2020-05-26 by damgaard22