wasm-micro-runtime/wasm-micro-runtime #264: Understanding WAMR AoT formats

Rochet2, 2020-05-26:
The AoT compiler can generate code that can run really fast. However there are multiple formats that the compiler can output. The formats are aot, object, llvmir-unopt and llvmir-opt.

AoT doesnt necessarily mean compiling to native code.
Is it right to assume that the `aot` format is optimized intermediate representation that WAMR can use, not native code?

wenyongh, 2020-05-27:
@Rochet2 currently the other three formats are for debug purpose only, iwasm cannot load and run them.
The aot format is refactored from the object file: WASM bytecode -> LLVM IR -> object file -> aot file. We extract the necessary data from object file, e.g. native code, relocation info, and extract the init data from wasm file, and then merge/re-org these data together to generate aot file.

Rochet2, 2020-05-27:
Is the native code able to break the sandbox?
For example, would it be able to edit the runtime or other modules run with the same runtime as they run in the same process?
Or are the function calls and memory access checked by the runtime prior/during run-time?

I am not talking about hypothetical breaches in the sandbox, just trying to figure out if there is any checking at all.

wenyongh, 2020-05-27:
Yes, the similar sandbox checks in interpreter are also applied in AoT native code, e.g. memory access boundary check, integer divide by zero check, call unlinked function check, stack overflow check and so on.

Rochet2, 2020-05-27:
Thanks a lot for clarifying.

closed 2020-05-27 by Rochet2