wasm-micro-runtime/wasm-micro-runtime #266: Binary size of AOT is not that good as what the Performance wiki shows

dpinthinker, 2020-05-30:
The Performance wiki:
https://github.com/bytecodealliance/wasm-micro-runtime/wiki/Performance#wasm-application-binary-size

It shows Native and AOT binary sizes are almost the same. 

Workload | GCC compiled native binary (KB) | Wasm bytecode (KB) | iwasm-aot module (KB)**
-- | -- | -- | --
coremark | 23120 | 10479 | 24496
base64 | 10224 | 6284 | 7236
gimli | 10224 | 905 | 2240

During my test, the binary size of AOT are 6~10x bigger than native. I just used the normal compile command.

```
wamrc -o test.aot test.wasm
```

Did I compile it in a wrong way?
Could you give some more details about how to compile the AOT correctly?

xwang98, 2020-05-30:
@dpinthinker Thank you for reporting the issue. Is it possible for you to provide the sample you tested? We would like to run the test in our environment.

dpinthinker, 2020-05-30:
@xwang98 You can try with fibonacci function, it works the same as my working codes.
https://www.tutorialspoint.com/data_structures_algorithms/fibonacci_recursive_program_in_c.htm

My test result on fibonacci function is following:

| main.c | a.out | main.wasm | main.aot | main-Os.wasm | main-Os.oat |
| ------ | ----- | --------- | -------- | ------------ | ----------- |
| 518    | 8672  | 20183     | 58932    | 19661        | 58160       |

Following is the command has been used.

a.out: 
```bash
gcc main.c
```

main.wasm/main.aot: 
```bash
/opt/wasi-sdk/bin/clang main.c -o main.wasm
wamrc -o main.aot main.wasm
```

main-Os.wasm/main-Os.aot: 
```
/opt/wasi-sdk/bin/clang -Os main.c -o main-Os.wasm
wamrc -o main-Os.aot 
```

wenyongh, 2020-05-31:
@dpinthinker it is mainly caused by wasi-sdk, for a.out generated by gcc, the libc symbols it depends on are external symbols and are implemented/stored in system libc, so normally the size is small. For wasi-sdk clang, it introduces [wasi-libc](https://github.com/WebAssembly/wasi-libc) and statically links wasi-libc together with app object file into app wasm bytecode, meaning that many extra libc source codes are compiled into app wasm file, so its size is large. For the fibonacci example, it calls printf, and the libc implementation of printf are compiled into wasm file.

WAMR provides libc-builtin to support a subset of libc APIs, and we can compile wasm app without **stdlib**, and run it with iwasm:

    /opt/wasi-sdk/bin/clang -o main.wasm main.c \
    -Wl,--export=main -Wl,--no-entry -nostdlib -Wl,--allow-undefined

Here is my test result in this mode:
| main.c | a.out | main.wasm | main.aot | main-Os.wasm | main-Os.oat |
| ------ | ----- | --------- | -------- | ------------ | ----------- |
| 503    | 8368  | 1049     | 2116 | 521| 1360|

dpinthinker, 2020-05-31:
@wenyongh Thank you! It is great!

Will the additional boundary and branch checking instructions for wasm cause aot binary size bigger when the application becomes much bigger?

wenyongh, 2020-05-31:
Yes, it makes aot binary size bigger and depends how many memory access instructions are generated, each boundary check might introduce several machine instructions. We have tried to refine the boundary check process, e.g. remove check if the address can be pre-calculated and is inside the memory address range, or remove duplicated check if there are two continuous accesses to the same address. Normally it increases the size about 20%~30% now.

dpinthinker, 2020-06-01:
It seems great. I will continue to test it. Thank you!

dpinthinker, 2020-06-01:
Linking crashed on libc symbol `CLOCK_PROCESS_CPUTIME_ID`, when compiling without libc.

```c
 #include <stdio.h>
 #include <time.h>
 int main()
 {
    struct timespec myTime;
    clock_gettime(CLOCK_PROCESS_CPUTIME_ID, &myTime);
    printf("current time: %lld\r\n", myTime.tv_sec);
    return 0;
 }
```

```
/opt/wasi-sdk/bin/clang -o main.wasm main.c \
-Wl,--export=main -Wl,--no-entry -nostdlib -Wl,--allow-undefined
wasm-ld: error: /var/folders/qh/wgh7bjmx2vzg8bssmwklftfh0000gn/T/main-9498d2.o: undefined symbol: _CLOCK_PROCESS_CPUTIME_ID
clang-10: error: linker command failed with exit code 1 (use -v to see invocation)
```

dpinthinker, 2020-06-02:
@wenyongh Do you know how to fix the `undefined symbol: _CLOCK_PROCESS_CPUTIME_ID` issue?

wenyongh, 2020-06-02:
@dpinthinker could you extend the native library to implement another version of clock_gettime? Ref to [export native api](https://github.com/bytecodealliance/wasm-micro-runtime/blob/master/doc/export_native_api.md). For example, implement and register a native API named my_clock_gettime(), which calls clock_gettime(CLOCK_PROCESS_CPUTIME_ID, &myTime), and then calls my_clock_gettime() in your wasm application.

dpinthinker, 2020-06-02:
@wenyongh This way is a little tricky. Could we move the timer.h to the runtime buildin-libc?

wenyongh, 2020-06-12:
Should be not tricky, sometimes we really need implement native API, similar to Java, JavaScript and other VMs. Moving timer.h to libc-builtin won't work, since it wasn't implemented and registered, runtime cannot find it the registered native API list and call it. Please ref to #273. Thanks.

closed 2020-06-12 by wenyongh