wasm-micro-runtime/wasm-micro-runtime #267: wamrc can not output AOT file normally on MacOS

dpinthinker, 2020-05-30:
Here is my test codes:
https://www.tutorialspoint.com/data_structures_algorithms/fibonacci_recursive_program_in_c.htm

```bash
➜  fib wamrc -o main.aot main.wasm
Create AoT compiler with:
  target:        x86_64
  target cpu:    skylake
  cpu features:
  opt level:     3
  size level:    3
  output format: AoT file
Error: invaid llvm binary bin_type.
```
It outputs an error `Error: invaid llvm binary bin_type.`. After reading codes, it seems failed at `aot_resolve_target_info`.

Caused in `aot_resolve_target_info`.
https://github.com/bytecodealliance/wasm-micro-runtime/blob/c182eebc6b66fd74bbae113556b6100068d0c8d0/core/iwasm/compilation/aot_emit_aot_file.c#L1388-L1400

Called by `aot_obj_data_create`.
https://github.com/bytecodealliance/wasm-micro-runtime/blob/c182eebc6b66fd74bbae113556b6100068d0c8d0/core/iwasm/compilation/aot_emit_aot_file.c#L1909-L1916

wenyongh, 2020-05-31:
@dpinthinker seems that the executable file format in MacOS is Mach-O, but not ELF, see [Comparison_of_executable_file_formats](https://en.wikipedia.org/wiki/Comparison_of_executable_file_formats). Could you please modify codes as:

    if (bin_type != LLVMBinaryTypeELF32L 
         && bin_type != LLVMBinaryTypeELF32B 
         && bin_type != LLVMBinaryTypeELF64L 
         && bin_type != LLVMBinaryTypeELF64B
         && bin_type != LLVMBinaryTypeMachO32L
         && bin_type != LLVMBinaryTypeMachO32B
         && bin_type != LLVMBinaryTypeMachO64L
         && bin_type != LLVMBinaryTypeMachO64B) { 
         aot_set_last_error("invaid llvm binary bin_type."); 
         return false; 
     }

and try again? thanks.

dpinthinker, 2020-05-31:
After add those Mach Types, it works!

closed 2020-05-31 by dpinthinker