wasm-micro-runtime/wasm-micro-runtime #270: Both wasm and AOT file can not run on MacOs.

dpinthinker, 2020-05-31:
Codes are just to print hello word string.
```c
 #include <stdio.h>
 int main()
 {
    printf("Hello, World!");
    return 0;
 }
```
wasm file crash Info:
```
wasm main.wasm
[17:45:31:051 - C4F8DC0]: warning: fail to link import function (wasi_snapshot_preview1, proc_exit)

[17:45:31:107 - C4F8DC0]: warning: fail to link import function (wasi_snapshot_preview1, fd_close)

[17:45:31:113 - C4F8DC0]: warning: fail to link import function (wasi_snapshot_preview1, fd_write)

[17:45:31:117 - C4F8DC0]: warning: fail to link import function (wasi_snapshot_preview1, fd_fdstat_get)

[17:45:31:123 - C4F8DC0]: warning: fail to link import function (wasi_snapshot_preview1, fd_seek)

[17:45:31:986 - C4F8DC0]: WASM execute application failed: main function not found.

Exception: lookup main function failed.
```

AOT file crash Info:
```
iwasm main.aot
AOT module load failed: invalid object file type, expected relocatable file type but got others.
```

wenyongh, 2020-05-31:
libc-wasi is disabled by default in darwin platform, could you please try enabling it:

    cmake .. -DWAMR_BUILD_LIBC_WASI=1

Or compile the C source code to wasm app without stdlib to disable wasi in wasm app:

    /opt/wasi-sdk/bin/clang -o main.wasm main.c \
    -Wl,--export=main -Wl,--no-entry -nostdlib -Wl,--allow-undefined

For AOT, since the object file format in MacOS is Mach-O but not ELF, we don't know whether some info parsed in the object file is as expected, could you try skip the related checks in aot_loader.c if it reports error? thanks.

dpinthinker, 2020-05-31:
@wenyongh 

For wasm:
I don't know which target to build with "cmake .. -DWAMR_BUILD_LIBC_WASI=1".
Using the "/opt/.../clang -o ..." command, the wasm file works.

For AOT:
I changed the code as following and recompiled iwasm.
```
diff --git a/core/iwasm/aot/aot_loader.c b/core/iwasm/aot/aot_loader.c
index 560030d..f943da6 100644
--- a/core/iwasm/aot/aot_loader.c
+++ b/core/iwasm/aot/aot_loader.c
@@ -300,12 +300,12 @@ load_target_info_section(const uint8 *buf, const uint8 *buf_end,
     }

     /* Check target elf file type */
-    if (target_info.e_type != E_TYPE_REL) {
-        set_error_buf(error_buf, error_buf_size,
-                      "AOT module load failed: invalid object file type, "
-                      "expected relocatable file type but got others.");
-        return false;
-    }
+    //if (target_info.e_type != E_TYPE_REL) {
+    //    set_error_buf(error_buf, error_buf_size,
+    //                  "AOT module load failed: invalid object file type, "
+    //                  "expected relocatable file type but got others.");
+    //    return false;
+    //}

     /* Check machine info */
     if (!check_machine_info(&target_info, error_buf, error_buf_size)) {
```

and rerun aot file, it crashed again.
```
➜  helloworld iwasm main.aot
AOT module load failed: unknown machine type 0.
```

dpinthinker, 2020-06-01:
It seems aot_loader.c doesn't support Mach-O types yet.

https://github.com/bytecodealliance/wasm-micro-runtime/blob/e81f72d41fa7b3b36783ad298041d1d0e28d10e1/core/iwasm/aot/aot_loader.c#L126-L148

wenyongh, 2020-06-02:
Yes, WAMR only supports ELF file format, and machine types listed in aot_loader.c. If you want to support Mach-O, you might need to enable it by yourself, or bypass some checks to see whether it works. Wonder why you need to enable wamrc in MacOS, could you please generate the AOT file in normal Linux platform e.g. Ubuntu, and then send it to MacOS to run?

dpinthinker, 2020-06-02:
I just want to compile the wasm on MacOs, and run the aot file on IOT devices finally. 
Try to run on MacOs is just a attempt to run example codes. I will try to run the aot file on Ubuntu VM.

keith2018, 2020-06-09:
"x86_64" works fine on MacOS

dpinthinker, 2020-06-09:
> "x86_64" works fine on MacOS

@keith2018  Could you give more details? Like the commands.

dpinthinker, 2020-07-03:
It works when set target as 'x86_64', as following.

```
wamrc --target=x86_64 -o main.aot main.wasm
```

closed 2020-07-03 by dpinthinker