wasm-micro-runtime/wasm-micro-runtime #272: I can not

victortanjh, 2020-06-02:


victortanjh, 2020-06-02:
if i call wasm_runtime_full_init  and wasm_runtime_load and wasm_runtime_instantiate for one module.
then i call a function in this module in another thread, but failed. it said:call wasm function [**] failed. Exception: unreachable

Weining2019, 2020-06-04:
Hi @victortanjh , I just make a quick test on Ubuntu to call the function in another thread. It shows the hello-world samples work well. Here is the patch. And could you share your detailed usage?

--- a/product-mini/platforms/linux/main.c
+++ b/product-mini/platforms/linux/main.c
@@ -9,6 +9,8 @@
 #include <stdlib.h>
 #include <string.h>
 
+#include <pthread.h>
+
 #include "bh_platform.h"
 #include "bh_read_file.h"
 #include "wasm_export.h"
@@ -364,12 +366,17 @@ main(int argc, char *argv[])
         goto fail3;
     }
 
+#if 0
     if (is_repl_mode)
         app_instance_repl(wasm_module_inst);
     else if (func_name)
         app_instance_func(wasm_module_inst, func_name);
     else
         app_instance_main(wasm_module_inst);
+#endif
+    pthread_t thread_id;
+    pthread_create(&thread_id, NULL, app_instance_main, wasm_module_inst);
+    sleep(10);

product-mini/app-samples/hello-world $ iwasm test.wasm 
Hello world!
buf ptr: 0xffffc2c8
buf: 1234

victortanjh, 2020-07-24:
thank you for your reply,  the reason for calling failed from another thread is that I forgot to set native_symbols to be static.

closed 2020-08-20 by wenyongh