wasm-micro-runtime/wasm-micro-runtime #273: What is the purpose of "Export native API to WASM application"?

dpinthinker, 2020-06-03:
As wasm are compiled from native codes like c/c++, who do WAMR still need to export native API?

Will this import so (shared library) to the wasm application?

If there is just not enough api in builtin-libc, can we copy more .h/.c files to builtin-libc?

wenyongh, 2020-06-03:
@dpinthinker though we can compile C/C++/typescript source codes to wasm file, we might still need native API to do something, there might be some operations related to platform and hardware, which are low level operations and cannot be implemented in application's source codes, e.g. File IO, network, time, sensor, graphics.

WAMR provides the mechanism to export native API, normally the native API's codes can be statically linked with runtime, and no need to import shared library.

You cannot just copy .h/.c files to libc-builtin, normally you should implement the native API wrappers by yourself, and then define them as symbol array, and register the array to runtime, please ref to the document and samples/basic.

dpinthinker, 2020-06-03:
@wenyongh Do the export native apis fit well into WebAssembly's sanbox model?

In [WebAssembly's WASI](https://github.com/WebAssembly/WASI/blob/master/docs/WASI-overview.md#capability-oriented), it shows their code design follows CloudABI's concept of capability-based security. 
Does the design of the exporting native api follow this concept also, and is it still safe?

wenyongh, 2020-06-04:
> @wenyongh Do the export native apis fit well into WebAssembly's sanbox model?

Yes, WAMR provides mechanism to 
  (1) auto-check the native API's pointer parameters when calling native API, developer can specify which parameter is pointer or string and specify the pointer's length, with letter '*', '~', '$', please ref to "Function signature" of [Export native API](https://github.com/bytecodealliance/wasm-micro-runtime/blob/master/doc/export_native_api.md) for more details
 (2) use APIs to do the address conversion between wasm app relative offset and native absolute address, and validate whether the address is in app's memory space, see [Buffer address conversion and boundary check](https://github.com/bytecodealliance/wasm-micro-runtime/blob/master/doc/export_native_api.md#buffer-address-conversion-and-boundary-check). For example, developer can validate the pointer parameter by himself. To see all the APIs, please ref to [wasm_export.h](https://github.com/bytecodealliance/wasm-micro-runtime/blob/master/core/iwasm/include/wasm_export.h)
 (3) allocate memory from app's heap space (app memory space includes app heap space and app linear memory space), and pass the memory buffer to wasm app.

> 
> In [WebAssembly's WASI](https://github.com/WebAssembly/WASI/blob/master/docs/WASI-overview.md#capability-oriented), it shows their code design follows CloudABI's concept of capability-based security.
> Does the design of the exporting native api also follow this concept, and is it still safe?

The WASI feature of WAMR follows the concept of capability-based security, you can run iwasm with --env=<env> and --dir=<dir> to specify the environment variables and the directories that app can access.
The capability-based security is about the file/dir operation, actually it hasn't matter with the design of exporting native api.

closed 2020-08-27 by wenyongh