wasm-micro-runtime/wasm-micro-runtime #281: Call specific function from wasm module inside Zephyr

xicocana, 2020-06-14:
I am running WAMR on zephyr inside QEMU.
I have been able to follow the sample available at: /product_mini/platforms/zephyr/sample, and I can now run the main function of wasm modules.

However, I am trying to run specific functions inside the same wasm module with no success.
I have changed the simple sample and instead of 'wasm_application_execute_main' I am calling 'wasm_application_execute_func'.

e.g. (using a wasm module with a single function "sum" that adds the two given arguments): 
in sample/src/main.c:
char* balance[2] = {"15","5"};
wasm_application_execute_func(module_inst, "sum", 2, balance);

I get a big amount of errors and the module doesn't run.
I do not have that many experience with C and I am not being able to fix this issue.
Is there any available similar sample somewhere that can help me?

I'm sorry if this is not the place to ask this,
Thanks in advance

wustwn, 2020-06-18:
@xicocana the wasm_runtime_execute_func is not supported in zephyr platform currently, the compiler reports link error while building. We recommend you use the API wasm_runtime_lookup_function and wasm_runtime_call_wasm instead, see the attachment patch for sample/simple/src/main.c. for the APIs, please refer to [Native calls WASM functions and passes parameters](https://github.com/bytecodealliance/wasm-micro-runtime/blob/master/doc/embed_wamr.md#native-calls-wasm-functions-and-passes-parameters). 

we wrote a wasm application named main.c:

    int sum(int x, int y)
    {
        return x + y;
    }

then compile it to test.wasm with wasi-sdk:

    /opt/wasi-sdk/bin/clang -O3 -z stack-size=4096 -Wl,--initial-memory=65536 \
          -Wl,--export=main -Wl,--export=sum -o test.wasm main.c \
          -Wl,--export=__heap_base,--export=__data_end -Wl,--no-entry \
          -nostdlib -Wl,--allow-undefined

then we dump it to test_wasm.h (posted attachment) and copy it to zephry/samples/simple/src/:

    cp test_wasm.h zephyr/samples/simple/src/

Finally, execute samples/simple/build_and_run.sh and get the correct return of sum function

    sum function return: 20

Could you please have a try? thanks.

[attachment.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/4795863/attachment.zip)

xicocana, 2020-06-18:
@wustwn It works perfectly now ! Thanks for the fast and detailed answer ! Really saved me some headaches here

closed 2020-06-18 by xicocana

xicocana, 2020-07-06:
@wustwn btw how are you dumping the .wasm to test-wasm.h? I wrote a small python script that calls hexdump and creates me the .h file with the desired syntax. Just wondering if there's a better way

wustwn, 2020-07-12:
@xicocana Hey there,  we have uploaded the ibinarydump tool we wrote, which you can use to dump .wasm file to test_wasm.h.

please refer to [binarydump-tool] (https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/test-tools/binarydump-tool)  and have a try.

xicocana, 2020-10-05:
@wustwn sorry for troubling you again.
I am successfuly being able to compile C code to wasm code using wasi-sdk, as you explained:
`/opt/wasi-sdk/bin/clang -O3 -z stack-size=4096 -Wl,--initial-memory=65536 \
      -Wl,--export=main -Wl,--export=sum -o test.wasm main.c \
      -Wl,--export=__heap_base,--export=__data_end -Wl,--no-entry \
      -nostdlib -Wl,--allow-undefined`

but everytime I try with a piece of C code with external libraries, e.g. : 

`#include <stdarg.h>`

I get the following WAMR error: 

`[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: cortex-a53
*** Booting Zephyr OS build zephyr-v2.2.0-2144-g3d16bda27d5e  ***
WASM module load failed: unexpected end
elpase: 10
QEMU: Terminated`

Should I compile my code in a different way?
Thanks in advance