wasm-micro-runtime/wasm-micro-runtime #286: Expose type information of WASM functions

Hexxeh, 2020-06-17:
When using the raw native export support, I can't see a way to determine the argument and return types of the function being called at runtime.

Similarly, when calling a WASM function from native, it doesn't appear to be possible to determine the argument/return types to validate the call we're about to make is valid.

This information exists in the WASM, so it seems it could be exposed for callers to validate. Not sure about the AoT implications though.

Is my understanding here correct, and if so, is this a valid addition? Any concerns?

xwang98, 2020-06-17:
@Hexxeh While calling WASM function from native, the function signature information is available in the runtime, but not exposed to the callers. We suppose the caller has the knowledge of the function signature when programming the code. 

When wasm code calls the native API, the function prototype should be declared in some header file for the WASM compilation. So it probably has no need to expose API for querying the native function signature.

If you have different usages that require such support, could you please explain a bit?

Hexxeh, 2020-06-18:
Similar use case to https://github.com/bytecodealliance/wasm-micro-runtime/issues/228: binding to another language like JavaScript and being able to read the type of an exported function to turn arguments into the correct types. Also, when calling into "native" from WASM, being able to determine the argument types to know how many arguments need to be converted/created.

Hexxeh, 2020-08-04:
It looks like this is now addressed via support for the wasm-c-api added in https://github.com/bytecodealliance/wasm-micro-runtime/pull/315, thanks!

closed 2020-08-04 by Hexxeh