wasm-micro-runtime/wasm-micro-runtime #289: Compiling error when export WAMR to Cortex-M4 board

oubotong, 2020-06-20:
When I try to export the WAMR to Cortex-M4 board I came across these errors:
```
../wamr/core/iwasm/common/arch/invokeNative_arm.s: Assembler messages:
../wamr/core/iwasm/common/arch/invokeNative_arm.s:19: Error: lo register required -- `stmfd sp!,{r4,r5,r6,r7,lr}'
../wamr/core/iwasm/common/arch/invokeNative_arm.s:29: Error: Thumb does not support this addressing mode -- `ldr r0,[r4],#4'
../wamr/core/iwasm/common/arch/invokeNative_arm.s:33: Error: Thumb does not support this addressing mode -- `ldr r1,[r4],#4'
../wamr/core/iwasm/common/arch/invokeNative_arm.s:37: Error: Thumb does not support this addressing mode -- `ldr r2,[r4],#4'
../wamr/core/iwasm/common/arch/invokeNative_arm.s:41: Error: Thumb does not support this addressing mode -- `ldr r3,[r4],#4'
```
And I solved this by adding `.syntax unified` in **invokeNative_arm.s** to let the assembler know I am using the unified assembly. With this modification, I am able to run WAMR on cortex-m4 board. I think this is not the best way to solve this issue and I hope someone could give out a better solution.

Weining2019, 2020-06-22:
The Cortex-M4 processor implements the ARMv7-M Thumb instruction set while `invokeNative_arm.s` is for ARM but not for Thumb. So you should use `invokeNative_thumb.s` instead. And if the board has FPU(and has been enabled) which means VFP registers s0-s15 (d0-d7, q0-q3) are used for passing arguments or returning results in standard procedure-call variants, you should use `invokeNative_thumb_vfp.s` instead.

If you build WAMR with CMake, just set the CMake variable `-DWAMR_BUILD_TARGET=THUMBV7` or  `-DWAMR_BUILD_TARGET=THUMBV7_VFP`.

I hope it helps.

oubotong, 2020-06-22:
Thanks, I will take a try

closed 2020-08-20 by wenyongh