wasm-micro-runtime/wasm-micro-runtime #295: support pass buffer directly between two modules

scottlin2019, 2020-07-01:
like this:

char *invokeModule(wasm_exec_env_t exec_env, uint32 *argv[], int argc)

xwang98, 2020-07-01:
@joshtriplett Is any spec proposal that might be related to this issue?

xwang98, 2020-07-01:
@abrown pointed this propsoal: 
https://github.com/WebAssembly/interface-types/blob/master/proposals/interface-types/Explainer.md#enabling-shared-nothing-linking-of-webassembly-modules

@scottlin2019

joshtriplett, 2020-07-01:
Passing a buffer between modules would definitely require some sort of memory-to-memory copying, at least at the moment, since modules wouldn't share a linear memory. That's part of what interface types should help to specify.

woodsmc, 2026-08-15:
I think this is a great feature request. In the 6 years (ish) since it was opened. There has been a lot of work on shared heaps for linear memory within WAMR that address this. I suggest we mark this as closed - we got there in the end.

Any objections?