wasm-micro-runtime/wasm-micro-runtime #298: Why does appmanager need to remove some sections after loading and init aot/bytecode module?

dpinthinker, 2020-07-01:
AOT sections removing:

https://github.com/bytecodealliance/wasm-micro-runtime/blob/1f6b589c12083df1f9337cc26d41e14f6224839a/core/app-mgr/app-manager/module_wasm_app.c#L627-L635

https://github.com/bytecodealliance/wasm-micro-runtime/blob/1f6b589c12083df1f9337cc26d41e14f6224839a/core/app-mgr/app-manager/module_wasm_app.c#L651-L654

Bytecode sections removing:

https://github.com/bytecodealliance/wasm-micro-runtime/blob/1f6b589c12083df1f9337cc26d41e14f6224839a/core/app-mgr/app-manager/module_wasm_app.c#L690-L704

https://github.com/bytecodealliance/wasm-micro-runtime/blob/1f6b589c12083df1f9337cc26d41e14f6224839a/core/app-mgr/app-manager/module_wasm_app.c#L721-L724

https://github.com/bytecodealliance/wasm-micro-runtime/blob/1f6b589c12083df1f9337cc26d41e14f6224839a/core/app-mgr/app-manager/module_wasm_app.c#L753-L756

While those sections are not be removed in the mini project.

xwang98, 2020-07-01:
@dpinthinker The intention here is to reduce the memory consumption.

dpinthinker, 2020-07-01:
Will this cause the lack of some WASM features?

xwang98, 2020-07-01:
No. After the wasm module is loaded, some sections have been coverted to internal data structures. there is no need to keep the orginal sections.

dpinthinker, 2020-07-01:
It seems great!

closed 2020-07-01 by dpinthinker