wasm-micro-runtime/wasm-micro-runtime #304: Compiling a c library to be used on linux platform

rahul-thakoor, 2020-07-07:
Hello,

I trying to compile libraries such as https://github.com/jackmitch/libsoc to be used on Linux hosts of different architectures. I have tried to use emscripten to create a standalone wasm module without linking libc which requires the following imports:

```
  (import "env" "sprintf" (func $env.sprintf (type $t0)))
  (import "env" "perror" (func $env.perror (type $t2)))
  (import "env" "malloc" (func $env.malloc (type $t1)))
  (import "env" "free" (func $env.free (type $t2)))
  (import "env" "open" (func $env.open (type $t0)))
  (import "env" "write" (func $env.write (type $t0)))
  (import "env" "access" (func $env.access (type $t3)))
  (import "env" "close" (func $env.close (type $t1)))
  (import "env" "emscripten_memcpy_big" (func $env.emscripten_memcpy_big (type $t0)))
  (import "env" "realloc" (func $env.realloc (type $t3)))
  (import "env" "setTempRet0" (func $env.setTempRet0 (type $t2)))
```
Does the builtin libc provide those definitions? when trying to run with iwasm I get

````
[03:26:58:610 - 42710740]: warning: fail to link import function (env, perror)

[03:26:58:660 - 42710740]: warning: fail to link import function (env, open)

[03:26:58:700 - 42710740]: warning: fail to link import function (env, write)

[03:26:58:708 - 42710740]: warning: fail to link import function (env, access)

[03:26:58:714 - 42710740]: warning: fail to link import function (env, close)

[03:26:58:744 - 42710740]: warning: fail to link import function (env, realloc)
````
Is it possible at all? maybe by compiling using the wasi-sdk clang and wamr-sdk built-in sysroot?

Thanks

rahul-thakoor, 2020-07-07:
if building the wasm module using emscripten and linking libc, only 

```
  (import "env" "__sys_access" (func $env.__sys_access (type $t2)))
```
needs to be imported

wenyongh, 2020-07-10:
@tyler Not sure what __sys_access is used for and whether it is called actually, but wasi_snapshot_preview1 is supported if building iwasm with libc-wasi support (cmake -DWAMR_BUILD_LIBC_WASI=1). Could you please try again?

rahul-thakoor, 2020-07-13:
hey @wenyongh thanks for getting back. I will try again and get back

rahul-thakoor, 2020-07-13:
Here is the error message when building the wasm module using emscripten with libc support:

```
[00:34:48:559 - DDFB0740]: warning: fail to link import function (env, __sys_access)
Exception: fail to call unlinked import function (env, __sys_access)
```

and if I build the wasm module with `EMCC_ONLY_FORCED_STDLIBS=1` as described [here](https://github.com/emscripten-core/emscripten/wiki/WebAssembly-Standalone#supplying-your-own-system-libraries) and run with iwasm, I get the following output:

```
[01:04:48:964 - CFA00740]: warning: fail to link import function (env, perror)
[01:04:48:093 - CFA00740]: warning: fail to link import function (env, siprintf)
[01:04:48:104 - CFA00740]: warning: fail to link import function (env, close)
[01:04:48:113 - CFA00740]: warning: fail to link import function (env, access)
[01:04:48:123 - CFA00740]: warning: fail to link import function (env, write)
[01:04:48:132 - CFA00740]: warning: fail to link import function (env, open)
Exception: fail to call unlinked import function (env, siprintf)
```

I built iwasm with ` -DWAMR_BUILD_LIBC_BUILTIN=1 -DWAMR_BUILD_LIBC_WASI=1`

Thanks

wenyongh, 2020-07-16:
Maybe you can build wasm module with libc support, and implement the native function __sys_access by yourself, and register the native API to WAMR. Please ref to [export native api](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/export_native_api.md).

For example, if the function type of __sys_access in built wasm app is (i32, i32)i32, you might implement the function like:

    int __sys_access_wrapper(wasm_exec_env_t exec_env, char *pathname, int mode)
    {
        return access(pathname, mode);
    }

Then define the native API array:

    static NativeSymbol native_symbols[] = {
        {  "__sys_access", __sys_access_wrapper, "($i)i" }
    };

And register it after calling wasm_runtime_init() or wasm_runtime_full_init():

    int n_native_symbols = sizeof(native_symbols) / sizeof(NativeSymbol);
    if (!wasm_runtime_register_natives("env", native_symbols, n_native_symbols)) {
        report error;
    }

I haven't tested the wrapper function, hope it works for you.

closed 2020-08-20 by wenyongh