wasm-micro-runtime/wasm-micro-runtime #310: Error about building wamrc AoT compiler

kk8916, 2020-07-16:
I can build a wamrc AoT compiler for Linux correctly. But when I use the command  ` cmake -DWAMR_BUILD_TARGET=ARM ..` to build the AOT compiler, some errors were reported. 

```
..core/iwasm/common/arch/invokeNative_arm.s: Assembler messages:
..core/iwasm/common/arch/invokeNative_arm.s:19: Error: no such instruction: 'stmfd sp!,{r4,r5,r6,r7,lr}' 
..core/iwasm/common/arch/invokeNative_arm.s:20: Error: too many memory references for 'mov'
..core/iwasm/common/arch/invokeNative_arm.s:21: Error: too many memory references for 'mov'
..core/iwasm/common/arch/invokeNative_arm.s:22: Error: too many memory references for 'mov'
..core/iwasm/common/arch/invokeNative_arm.s:24: Error: expecting operand after ','; got nothing
```
I used the the command  ` cmake -DWAMR_BUILD_TARGET=AARCH64 ..` and similar errors happened in `invokeNative_aarch64.s`. I see WAMR_BUILD_TARGET values support  "X86_64", "AMD_64", "X86_32", "ARM_32", "MIPS_32", "XTENSA_32".What should I do to build a wamrc AoT compiler for ARM ?

Thanks in advance!

Weining2019, 2020-07-16:
Hi, the architecture of your host environment is ARM? Not x86?

kk8916, 2020-07-16:
> Hi, the architecture of your host environment is ARM? Not x86?

I'm building for ARM architecture on a X86 development machine.

Weining2019, 2020-07-16:
Why do you need to build wamrc for ARM? Actually you can build it for the development machine and pass --target=arm to wamrc.

For example:
./wamrc --target=armv7 -o test.aot test.wasm

kk8916, 2020-07-16:
> Why do you need to build wamrc for ARM? Actually you can build it for the development machine and pass --target=arm to wamrc.
> 
> For example:
> ./wamrc --target=armv7 -o test.aot test.wasm

I know that way to generate an AOT file. But I still want to try to build wamrc for ARM.I want to run the wamrc in other platform,such as phone.

Weining2019, 2020-07-16:
OK, I understand your scenario.

From the erro log:
..core/iwasm/common/arch/invokeNative_arm.s: Assembler messages:
..core/iwasm/common/arch/invokeNative_arm.s:19: Error: no such instruction: 'stmfd sp!,{r4,r5,r6,r7,lr}' 
..core/iwasm/common/arch/invokeNative_arm.s:20: Error: too many memory references for 'mov'
..core/iwasm/common/arch/invokeNative_arm.s:21: Error: too many memory references for 'mov'
..core/iwasm/common/arch/invokeNative_arm.s:22: Error: too many memory references for 'mov'
..core/iwasm/common/arch/invokeNative_arm.s:24: Error: expecting operand after ','; got nothing

Seem that you're using 'thumb' mode? What cross-toolchain do you use? Did you pass a '-mthumb' option to the compiler? Or this option is enabled by default?

Weining2019, 2020-07-16:
If you are targeting thumb, you should define -DWAMR_BUILD_TARGET=THUMB when build wamrc. Then core/iwasm/common/arch/invokeNative_thumb.s will be used instead of core/iwasm/common/arch/invokeNative_arm.s

kk8916, 2020-07-17:
> If you are targeting thumb, you should define -DWAMR_BUILD_TARGET=THUMB when build wamrc. Then core/iwasm/common/arch/invokeNative_thumb.s will be used instead of core/iwasm/common/arch/invokeNative_arm.s

Sorry, it seems like I forgot to set the CMAKE_TOOLCHAIN_FILE option.My cross-toolchains are `arm-linux-gnueabihf-gcc` and `arm-linux-gnueabihf-g++`. Then I build the wamrc AOT compiler again. The error was like below.
```
No rule to make target 'wasm-micro-runtime/core/deps/llvm/build/lib/libLLVMObjCARCOpts.a', needed by 'wamrc'. Stop
```
I think this error was caused by not recompiling the llvm using cross-toolchain. Then I imitate the build_llvm.sh and add the CMAKE_TOOLCHAIN_FILE option.The error was like below.
```
/home/kk/wamr-2/wasm-micro-runtime/core/deps/llvm/llvm/lib/Transforms/Hello/Hello.cpp:26: error: undefined reference to 'llvm::Pass::~Pass()'
/home/kk/wamr-2/wasm-micro-runtime/core/deps/llvm/llvm/lib/Transforms/Hello/Hello.cpp:32: error: undefined reference to 'llvm::errs()'
/home/kk/wamr-2/wasm-micro-runtime/core/deps/llvm/llvm/lib/Transforms/Hello/Hello.cpp:33: error: undefined reference to 'llvm::errs()'
/home/kk/wamr-2/wasm-micro-runtime/core/deps/llvm/llvm/lib/Transforms/Hello/Hello.cpp:33: error: undefined reference to 'llvm::Value::getName() const'
/home/kk/wamr-2/wasm-micro-runtime/core/deps/llvm/llvm/lib/Transforms/Hello/Hello.cpp:33: error: undefined reference to 'llvm::raw_ostream::write_escaped(llvm::StringRef, bool)'
/home/kk/wamr-2/wasm-micro-runtime/core/deps/llvm/llvm/include/llvm/Support/raw_ostream.h:162: error: undefined reference to 'llvm::raw_ostream::write(unsigned char)'
/home/kk/wamr-2/wasm-micro-runtime/core/deps/llvm/llvm/include/llvm/Support/raw_ostream.h:187: error: undefined reference to 'llvm::raw_ostream::write(char const*, unsigned int)'
/home/kk/wamr-2/wasm-micro-runtime/core/deps/llvm/llvm/lib/Transforms/Hello/Hello.cpp:44: error: undefined reference to 'llvm::Pass::~Pass()'
/home/kk/wamr-2/wasm-micro-runtime/core/deps/llvm/llvm/lib/Transforms/Hello/Hello.cpp:50: error: undefined reference to 'llvm::errs()'
/home/kk/wamr-2/wasm-micro-runtime/core/deps/llvm/llvm/lib/Transforms/Hello/Hello.cpp:51: error: undefined reference to 'llvm::errs()'
/home/kk/wamr-2/wasm-micro-runtime/core/deps/llvm/llvm/lib/Transforms/Hello/Hello.cpp:51: error: undefined reference to 'llvm::Value::getName() const'
/home/kk/wamr-2/wasm-micro-runtime/core/deps/llvm/llvm/lib/Transforms/Hello/Hello.cpp:51: error: undefined reference to 'llvm::raw_ostream::write_escaped(llvm::StringRef, bool)'
/home/kk/wamr-2/wasm-micro-runtime/core/deps/llvm/llvm/include/llvm/PassSupport.h:101: error: undefined reference to 'llvm::PassRegistry::getPassRegistry()'
/home/kk/wamr-2/wasm-micro-runtime/core/deps/llvm/llvm/include/llvm/PassSupport.h:101: error: undefined reference to 'llvm::PassRegistry::registerPass(llvm::PassInfo const&, bool)'
/home/kk/wamr-2/wasm-micro-runtime/core/deps/llvm/llvm/include/llvm/PassSupport.h:101: error: undefined reference to 'llvm::PassRegistry::getPassRegistry()'
/home/kk/wamr-2/wasm-micro-runtime/core/deps/llvm/llvm/include/llvm/PassSupport.h:101: error: undefined reference to 'llvm::PassRegistry::registerPass(llvm::PassInfo const&, bool)'
CMakeFiles/LLVMHello.dir/Hello.cpp.o:Hello.cpp:vtable for (anonymous namespace)::Hello: error: undefined reference to 'llvm::Pass::~Pass()'
CMakeFiles/LLVMHello.dir/Hello.cpp.o:Hello.cpp:vtable for (anonymous namespace)::Hello: error: undefined reference to 'llvm::Pass::getPassName() const'
CMakeFiles/LLVMHello.dir/Hello.cpp.o:Hello.cpp:vtable for (anonymous namespace)::Hello: error: undefined reference to 'llvm::Pass::print(llvm::raw_ostream&, llvm::Module const*) const'
CMakeFiles/LLVMHello.dir/Hello.cpp.o:Hello.cpp:vtable for (anonymous namespace)::Hello: error: undefined reference to 'llvm::FunctionPass::createPrinterPass(llvm::raw_ostream&, std::__ndk1::basic_string<char, std::__ndk1::char_traits<char>, std::__ndk1::allocator<char> > const&) const'
CMakeFiles/LLVMHello.dir/Hello.cpp.o:Hello.cpp:vtable for (anonymous namespace)::Hello: error: undefined reference to 'llvm::FunctionPass::assignPassManager(llvm::PMStack&, llvm::PassManagerType)'
CMakeFiles/LLVMHello.dir/Hello.cpp.o:Hello.cpp:vtable for (anonymous namespace)::Hello: error: undefined reference to 'llvm::Pass::preparePassManager(llvm::PMStack&)'
CMakeFiles/LLVMHello.dir/Hello.cpp.o:Hello.cpp:vtable for (anonymous namespace)::Hello: error: undefined reference to 'llvm::FunctionPass::getPotentialPassManagerType() const'
CMakeFiles/LLVMHello.dir/Hello.cpp.o:Hello.cpp:vtable for (anonymous namespace)::Hello: error: undefined reference to 'llvm::Pass::getAnalysisUsage(llvm::AnalysisUsage&) const'
CMakeFiles/LLVMHello.dir/Hello.cpp.o:Hello.cpp:vtable for (anonymous namespace)::Hello: error: undefined reference to 'llvm::Pass::releaseMemory()'
CMakeFiles/LLVMHello.dir/Hello.cpp.o:Hello.cpp:vtable for (anonymous namespace)::Hello: error: undefined reference to 'llvm::Pass::getAdjustedAnalysisPointer(void const*)'
CMakeFiles/LLVMHello.dir/Hello.cpp.o:Hello.cpp:vtable for (anonymous namespace)::Hello: error: undefined reference to 'llvm::Pass::getAsImmutablePass()'
CMakeFiles/LLVMHello.dir/Hello.cpp.o:Hello.cpp:vtable for (anonymous namespace)::Hello: error: undefined reference to 'llvm::Pass::getAsPMDataManager()'
CMakeFiles/LLVMHello.dir/Hello.cpp.o:Hello.cpp:vtable for (anonymous namespace)::Hello: error: undefined reference to 'llvm::Pass::verifyAnalysis() const'
[... cut ...]
CMakeFiles/LLVMHello.dir/Hello.cpp.o:Hello.cpp:vtable for (anonymous namespace)::Hello2: error: undefined reference to 'llvm::Pass::verifyAnalysis() const'
CMakeFiles/LLVMHello.dir/Hello.cpp.o:Hello.cpp:vtable for (anonymous namespace)::Hello2: error: undefined reference to 'llvm::Pass::dumpPassStructure(unsigned int)'
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
lib/Transforms/Hello/CMakeFiles/LLVMHello.dir/build.make:103: recipe for target 'lib/LLVMHello.so' failed

```
Did I miss something to set ? Please give me some suggestion. Thanks in advance!

Weining2019, 2020-07-17:
Yes, you're right that LLVM should be rebuilt by cross-toolchain. But we haven't tried before. Maybe [this](https://llvm.org/docs/HowToCrossCompileLLVM.html) can help.

@wenyongh Do you think it is feasible to cross-compile wamrc (and LLVM)?

wenyongh, 2020-07-28:
@kk8916 not sure what causes the linkage error, could you please use "make VERBOSE=1" to show the detailed link command? Make sure that the LLVM libraires are linked, e.g. :

    ${WAMR_ROOT}/core/deps/llvm/build/lib/libLLVMDemangle.a
    ${WAMR_ROOT}/core/deps/llvm/build/lib/libLLVMSupport.a
    ${WAMR_ROOT}/core/deps/llvm/build/lib/libLLVMTableGen.a
    ${WAMR_ROOT}/core/deps/llvm/build/lib/libLLVMCore.a
     ...
There should be a lot of libraries to link. Another reason might be building with threads, not sure whether "make -j n" of your cross compiler works correctly, e.g. thread A hasn't output the .a library, while thread B is linking it? Could you please try just typing make?

And if it doesn't work, try to find which library exports these symbols in llvm/build/lib folder, and link it manually.

Hope it helps.

closed 2020-08-28 by wenyongh