wasm-micro-runtime/wasm-micro-runtime #312: Export native function failed with return type 'string'

keith2018, 2020-07-17:
Hi，my native function is

```
const char* getStringProp(int id);
```

binding:

```
EXPORT_WASM_API_WITH_ATT(getStringProp, "(i)$", ctx)
```

it failed when WASM code calls 'getStringProp'

dpinthinker, 2020-07-18:
It seems that your signature is invalid.

https://github.com/bytecodealliance/wasm-micro-runtime/blob/32b2943369bccce52fbbc86bc9239757d37cbf52/core/iwasm/common/wasm_native.c#L177-L183

dpinthinker, 2020-07-18:
According to [Define the native API](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/export_native_api.md#step-2-define-the-native-api), you should add `wasm_exec_env_t exec_env` as your native method's first argument.
 
```
int foo_native(wasm_exec_env_t exec_env , int a, int b)
{
    return a+b;
}

void foo2(wasm_exec_env_t exec_env, char * msg, uint8 * buffer, int buf_len)
{
    strncpy(buffer, msg, buf_len);
}
```

wenyongh, 2020-07-18:
@keith2018 currently the signature character '$', '*' and '~' are only available for function arguments, but not return value, so for your sample, please define the signature as "($$)i", and use wasm_runtime_addr_native_to_app() to convert the string to int32 offset, and return it, for example:

    int _getStringProp(wasm_exec_env_t exec_env, const char *propId, const char *defaultVal)
    {
        wasm_module_inst_t module_inst = wasm_runtime_get_module_inst(exec_env);
        char *ret_str = ...;
        int ret_offset = wasm_runtime_addr_native_to_app(module_inst, ret_str);
        return ret_offset;
    }

keith2018, 2020-07-20:
> @keith2018 currently the signature character '$', '*' and '~' are only available for function arguments, but not return value, so for your sample, please define the signature as "($$)i", and use [...]

there's no runtime error again, but cannot get the right result string in wasm code, my test case:

```
int _getStringProp(wasm_exec_env_t exec_env, const char *propId, const char *defaultVal) {
    GET_SYSTEM_CONTEXT(ctx)
    wasm_module_inst_t module_inst = wasm_runtime_get_module_inst(exec_env);

    static  char *retStr = "test_string";
    return wasm_runtime_addr_native_to_app(module_inst, (void*)retStr);
}
```

```
EXPORT_WASM_API_WITH_ATT(_getStringProp, "($$)i", systemBind),
```

C code compiled to wasm: 
```
const char* retStr = _getStringProp("/runtime/uri", "default_string");
printf("WASM: %s", retStr);
```

it only prints "WASM:"，shows that we got empty result string

wenyongh, 2020-07-20:
> int _getStringProp(wasm_exec_env_t exec_env, const char *propId, const char *defaultVal) {
>     GET_SYSTEM_CONTEXT(ctx)
>     wasm_module_inst_t module_inst = [...]

@keith2018 WebAssembly sandbox specifies its own memory space and runtime can only access data in the memory, so please don't pass string or pointer of native to runtime.  You might use wasm_runtime_module_malloc() to allocate memory from app heap of wasm memory space, store data to it and return offset to runtime, for example:

    char *ret_str;
    int32 ret_offset;
    uint32 size = 20;

    ret_offset = wasm_runtime_module_malloc(module_inst, size, (void **)&ret_str);
    if (ret_str != NULL) {
        snprintf(ret_str, size, "%s", "test_string");
    }
    return ret_offset;

And remember that the string should be freed by calling wasm_runtime_module_free() in some place, or the app heap might be exhausted.

keith2018, 2020-07-20:
@wenyongh 
thanks, it works now!

 i have exported 'wasm_runtime_module_free' as an native function

```
void _memFree(wasm_exec_env_t exec_env, int ptr) {
    wasm_module_inst_t module_inst = wasm_runtime_get_module_inst(exec_env);
    wasm_runtime_module_free(module_inst, ptr);
}
```

and call it from wasm code
```
char* retStr = _getStringProp("/runtime/uri", "default_string");
printf("WASM: %s", retStr);
_memFree(retStr);
```

closed 2020-07-20 by keith2018