wasm-micro-runtime/wasm-micro-runtime #317: Can we call wasm func in multi thread?

victortanjh, 2020-07-24:
we got a crash when calling : wasm_runtime_call_wasm(exec_env, func, 2, argvIn))  in multi thread.

For the same module_inst and the same func in wasm, can we call it in multi thread?

xujuntwt95329, 2020-07-27:
Hi,

I'm afraid you can't run functions in the same instance in multiple threads. The wasm linear memory may be grown during execution, which may change the base address of the memory space. If the memory is grown in thread A and accessed in thread B, then you may access the invalid memory area.

It seems that our `pthread library` is suitable for your situation. You can use `pthread_create` in your wasm application to create multiple threads, and you should compile your application source code with `-pthread` support in wasi-sdk, please refer to [pthread_library](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/pthread_library.md) for more details.

There is also a sample for [multi-thread](https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/samples/multi-thread).

Hope this can help you

victortanjh, 2020-07-30:
thank you for your reply.
I want to make multi thread in the native world, not in the wasm app and call  wasm_runtime_call_wasm with the same parameters in each thread.

xujuntwt95329, 2020-07-30:
We are evaluating this kind of usage internally, I'll update the information here once it is finished.
Thanks

xujuntwt95329, 2020-08-04:
@victortanjh 

We've provide some spawn APIs to run wasm functions in multi threads.

please refer to [embed_wamr guide](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/embed_wamr.md#execute-wasm-functions-in-multiple-threads) to see the usage.

[Here](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/samples/spawn-thread/src/main.c) is also a sample code for reference.

Hope this helps, and thanks for reporting this issue.

closed 2020-09-24 by wenyongh