wasm-micro-runtime/wasm-micro-runtime #318: How to utilize WAMR with SGX in my own .wasm app？

leyao-daily, 2020-07-27:
For example, I have a rust app and compile to .wasm.
I want run this .wasm in the SGX enclave.
I find the mini-product linux-sgx and samples in it.
But how can I run my app in SGX enclave with WAMR runtime?
Is there any guide or examples? Thanks.

leyao-daily, 2020-07-27:
And in warmc, I found -sgx feature, it describe "generate SGX code", what doest it mean? Is that means, when i exec iwasm xxx.aot, it will run in SGX enclave?

wenyongh, 2020-07-27:
@leyao-daily currently the wasm file is statically linked to runtime by byte array, you can replace file product-mini/platforms/linux-sgx/enclave-sample/Enclave/test_wasm.h with your own file by using binarydump to generate it:

    cd test-tools/binarydump-tool
    mkdir build && cd build
    cmake .. && make
    ./binarydump -o test_wasm.h -n wasm_test_file test.wasm
    (or ./binarydump -o test_wasm.h -n wasm_test_file test.aot)
    remove "__aligned(4)" in test_wasm.h
    cp -a test_wasm.h ${WAMR_ROOT}/product-mini/platforms/linux-sgx/enclave-sample/Enclave/
    cd ${WAMR_ROOT}/product-mini/platforms/linux-sgx/enclave-sample
    make clean && make

We have basically implemented running wasm file in host by sending it from host to enclave to execute with sgx ecall, we will submit the patch soon.

For the AoT, yes, wamr supports running aot file in SGX, by default it is disabled, you can use cmake -DWAMR_ENABLE_AOT=1 to enable it when building linux-sgx:

    cd ${WAMR_ROOT}/product-mini/platforms/linux-sgx/
    mkdir build && cd build
    cmake .. -DWAMR_BUILD_AOT=1

leyao-daily, 2020-07-28:
@wenyongh It is very helpful, Thanks a lot. 👍

wenyongh, 2020-07-28:
Welcome @leyao-daily , we have submitted a new patch, now the enclave sample is built as binary iwasm, which can run host wasm file or aot file (send to enclave to run), similar to iwasm built in Linux platform, please pull and try again.

leyao-daily, 2020-07-28:
Excellent! I have tried right now and it works well. 
Another question: I complie .wasm to aot file with warmc, and one with no -sgx and one with warnc -sgx -o xx.aot xx.wasm
Both of them can be run by linux-sgx iwasm. 
So what is the usage of the -sgx in warmc command. It describe "generate SGX code". What is that mean, what code it generates? Thanks again.

wenyongh, 2020-07-28:
Please had better use -sgx option to generate aot code for SGX platform, or llvm codegen might generate machine code with unsupported relocation type for SGX, and we cannot run that code in SGX.

closed 2020-10-30 by wenyongh