wasm-micro-runtime/wasm-micro-runtime #321: cannot enable AOT on Android

ifels, 2020-07-28:
when I use WAMR_BUILD_AOT=1 to enable AOT for android platform.
`cmake .. -DWAMR_BUILD_AOT=1`

the output is :
`-- Build Configurations:
     Build as target X86_32
     CMAKE_BUILD_TYPE Release
     WAMR Interpreter enabled
     WAMR AOT disabled
     WAMR JIT disabled
     Libc builtin enabled
     Libc WASI disabled
     Fast interpreter disabled
     Multiple modules disabled
-- Configuring done
-- Generating done
`

Rochet2, 2020-07-28:
The android cmake file explicitly sets the configurations.
https://github.com/bytecodealliance/wasm-micro-runtime/blob/88af12501d60ce15474196b482ecdc7f48df9576/product-mini/platforms/android/CMakeLists.txt#L21

The guide to compiling for android mentions this: https://github.com/bytecodealliance/wasm-micro-runtime/blob/ee315e404971f8d242603995c5837a23542472a9/doc/build_wamr.md#android
> Any compiling requirement should be satisfied by modifying product-mini/platforms/android/CMakeList.txt

However, I would say that it would be nice to have the settings configurable without editing the cmakelists file.
Maybe cmake options can be used, which I believe have the same effect as SET command in cmake. https://cmake.org/cmake/help/latest/command/option.html

Rochet2, 2020-07-28:
Additionally, for me it took a while to figure out that the wasm-jni.cpp is not configured to be a part of the example by default. You need to define for example `-DWAMR_RUNTIME_LIB_SOURCE=wasm-jni.cpp` so that the file is included here https://github.com/bytecodealliance/wasm-micro-runtime/blob/88af12501d60ce15474196b482ecdc7f48df9576/product-mini/platforms/android/CMakeLists.txt#L93
Took me a while to figure out.

ifels, 2020-08-28:
> The android cmake file explicitly sets the configurations.
> https://github.com/bytecodealliance/wasm-micro-runtime/blob/88af12501d60ce15474196b482ecdc7f48df9576/product-mini/platforms/android/CMakeLists.txt#L21
> 
> The guide to compiling for android mentions this: https://github.com/bytecodealliance/wasm-micro-runtime/blob/ee315e404971f8d242603995c5837a23542472a9/doc/build_wamr.md#android
> 
> > Any compiling requirement should be satisfied by modifying product-mini/platforms/android/CMakeList.txt
> 
> However, I would say that it would be nice to have the settings configurable without editing the cmakelists file.
> Same approach as here
> 
> https://github.com/bytecodealliance/wasm-micro-runtime/blob/88af12501d60ce15474196b482ecdc7f48df9576/product-mini/platforms/linux/CMakeLists.txt#L53-L56
> 
> 
> Or maybe cmake options can be used, which I believe have the same effect. https://cmake.org/cmake/help/latest/command/option.html

thanks, I can enable AOT now  by set WAMR_BUILD_AOT to 1 in  CMakeLists.txt.

closed 2020-08-28 by ifels