wasm-micro-runtime/wasm-micro-runtime #322: Build error: Building samples/basic for AARCH64 target

xicocana, 2020-07-28:
I am trying to build the basic sample for aarch64 target.

On the CMakeLists.txt I have changed the WAMR_BUILD_TARGET from:
```
set (WAMR_BUILD_TARGET "X86_64")
```
to:
```
set (WAMR_BUILD_TARGET "AARCH64")
```

but i get the following build error (log.txt file in attachment)
[log.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/4991650/log.txt)

xujuntwt95329, 2020-07-29:
Seems your toolchain doesn't support `aarch64` instructions, have you changed the `CMAKE_C_COMPILER` to your cross-compiler ?
You can use `make VERBOSE=1` to check which compiler you are using.

xicocana, 2020-07-29:
thanks @xujuntwt95329 ! I indeed forgot about the toolchain. I'll give it a try and report back

xicocana, 2020-07-30:
I changed the CMAKE_C_COMPILER to my cross compiler toolchain ```aarch64-none-elf-gcc```.
But now I'm having the following problem:
```
aarch64-none-elf-gcc: error: unrecognized command line option '-mindirect-branch-register'

```

Can I remove this command line option so that my toolchain works? Or is it essential? If that's the case could you recommend a different toolchain?

xicocana, 2020-07-30:
@xujuntwt95329 I have changed the cross compiler toolchain to ```aarch64-none-linux-gnu-gcc``` and all is working fine now. thanks

closed 2020-07-30 by xicocana