wasm-micro-runtime/wasm-micro-runtime #325: CMake issues with zlib

sergiizubchevskyy, 2020-07-31:
Good morning!

I was having issues with CMake, during LLVM build. 

Basically in order to proceed I needed to add the following line in the CMakeList.txt:
`find_package(ZLIB REQUIRED)`

CMake version I'm using right now is 3.18.0, and the one that is required is 2.8. 

I think this may cause issues on more modern CMakes.

wenyongh, 2020-07-31:
@sergiizubchevskyy seems that zlib isn't installed, could you please install it? Try whether the command works:

    sudo apt-get install zlib1g-dev

sergiizubchevskyy, 2020-07-31:
At the time of the issue I tried this command, it was already installed :smile: 

If it works on your machine with updated CMake and without adding find_package(ZLIB required), please close the issue.

Might be problems on my side, but as I sad, just adding find_package solved it.

wenyongh, 2020-07-31:
Could you try adding the following line to build_llvm.sh:

    -DLLVM_ENABLE_ZLIB:BOOL=OFF

sergiizubchevskyy, 2020-07-31:
At the moment I cannot recompile LLVM due to time constraints, sorry

But with the updated version of CMake you don't have any issues?

sergiizubchevskyy, 2020-07-31:
Actually managed to build LLVM again.

Your suggestion solved the issue!

Is this option should be explicitly disabled as you suggested at your main branch?

wenyongh, 2020-08-04:
Some options including this were disabled in latest patch to decrease the binary size of llvm library.

closed 2020-08-06 by wenyongh