wasm-micro-runtime/wasm-micro-runtime #327: Wierd stack behaviour reported by Valgrind Massif

sergiizubchevskyy, 2020-07-31:
Hello!

I'm currently benchmarking the WAMR, and came by some strange behavior observed with Massif.

When running in JIT mode, with Massif stack analysis disabled I've got the following results:
[image]

When running in AOT mode, with Massif stack analysis disabled I've got the following results:
[image]

However if I enable Massif stack analysis and execute again the WAMR in AOT mode I'll get the following warning:
[image]

And the results are a bit strange:
[image]

Could you explain the reason why from 96.4 kiB WAMR jumps to 8 MiB in the same AOT mode? Is this a known issue? Any of this results you consider wrong?

Thank you in advance.

wenyongh, 2020-08-01:
@sergiizubchevskyy WAMR uses hardware trap to do native stack overflow check, which sets several guard pages (set them as not readable and writable) near the stack boundary, and needs to touch each stack pages since the OS may lazily grow the stack mapping. The default native stack size of main thread is 8M, if you use valgrind with Massif stack analysis disabled, the native stack pages are not calculated; if Massif stack analysis is enabled, they are calculated.

The warning you got should be also caused by touching stack pages or setting stack page property.

sergiizubchevskyy, 2020-08-01:
Thank you for explanation @wenyongh.

Is  there a way to manipulate this stack size, more specifically to reduce it? 

For our use case the RAM usage is an important metric, I know that in the interpreter modes the RAM usage is low, but CPU performance is not quite acceptable.

wenyongh, 2020-08-01:
Yes, normally when getting the native stack boundary, we can set the return value to a place that is relatively near to the stack bottom, so the stack overflow check with hardware trap mechanism will touch less pages. Could you please test the patch uploaded? The value of APP_THREAD_STACK_SIZE_MAX is 256K, you can set it to other size, which must be a multiple of 4K.

BTW: it is interesting to view the memory consumption results with graphics showing, how to use valgrind tool to generate it?

[stack_boundary.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/5010493/stack_boundary.txt)

sergiizubchevskyy, 2020-08-01:
Will try your patch right now.

PS: I use the following command to generate output file:

bash
valgrind --tool=massif --time-unit=B --stacks=yes --massif-out-file=massif.out.1.wamr_aot_B_stacks_aot iwasm coremark-wasi.aot
```
* Time unit is set to 'B' so that Massif's graphic uses allocated bytes as a time-unit, instead of time or number of instructions executed
* Stack tracing enabled
That command shall produce file called "massif.out.1.wamr_aot_B_stacks_aot" which I open with Massif-Visualizer which I installed from Ubuntu Software Center, there are two programs with similar names, but I use the one with the capital 'M' and 'V' is the name. Problem is that it requires "massif.out.#" to be at the beginning of the name to work, that is why I provide custom name.

sergiizubchevskyy, 2020-08-01:
Your patch improved the RAM performance quite a lot:

[image]

Now at worst case it consumes 583,5 kiB! After the peak consumes arround 336,5 kiB

Also it did not affected the CoreMark CPU performance.

By any chance do you foresee any possible issues that this patch could provide to the runtime's security or performance?

wenyongh, 2020-08-02:
@sergiizubchevskyy there should be no other issues of security or performance introduced by this patch, is only changes the max available stack size, and normally 256KB is large enough to run most cases, you can even reduce it to 128KB or 64KB to have a try.

sergiizubchevskyy, 2020-08-03:
Sorry to bother but why your patch is not already a part of the source code? 

The RAM usage improves a lot with it  :+1:

wenyongh, 2020-08-11:
Yes, we just want to let developer use the default maximum stack size if the system supports it. Thanks for reporting the issue, the memory consumption is really important, we have uploaded patch #346 , and now you can set the max stack size with cmake -DWAMR_APP_THREAD_STACK_SIZE_MAX=n, could you please try again?

sergiizubchevskyy, 2020-08-11:
Sure, should I go with 256kB?

sergiizubchevskyy, 2020-08-11:
Sure, I went with 128kiB and got the following results:

With AOT:
[image]
with peak usage of 456,5 kiB

With JIT:
[image]
with peak usage of 11.6 MiB

I think this topic can be closed since patch #346 was created and merged.

Thank you for your time and clarifications :+1:

closed 2020-08-11 by sergiizubchevskyy