wasm-micro-runtime/wasm-micro-runtime #328: Performance issues with hardware trap in AoT

ZenLulz, 2020-07-31:
Hello !

I'm dealing with a performance issue that started in the commit [ee315e404971f8d242603995c5837a23542472a9](https://github.com/bytecodealliance/wasm-micro-runtime/commit/ee315e404971f8d242603995c5837a23542472a9). From that commit, I can see a lot of changes have been performed, with notably the introduction of hardware traps support for 64-bit. The performance issue I'm experiencing only concerns 64-bit app (and not SGX), which is the target of that new feature.

As an example of my personal benchmark, I get the following results when the HW trap are enabled (the default option now):

[image]

Interestingly, the performance of the same code is way faster in SGX compared to the one outside.

If I compile the runtime with the flag `set (WAMR_DISABLE_HW_BOUND_CHECK 1)`, I get the following result.

[image]

Which is what I got before the introduction of that feature, with a little overhead.

I don't know if I missed something, but this feature slows down my results.

wenyongh, 2020-08-01:
@ZenLulz thanks for reporting the issue! it is really interesting, not sure how you use the WAMR APIs and how you run your benchmark? Do you build iwasm form product-mini/platforms, or embed WAMR APIs into your own application? For example, do you use iwasm to run the wasm app's main function one time, or continuously calling wasm_runtime_call_wasm to call wasm function?

From our test, the patch improves performance a lot (by 10% to 30%), since if we use hardware trap to do boundary check and native stack overflow check, we don't need to generate and execute the checking instructions in AOT code. The hardware trap will be trigged when aot code accesses the guard pages, and runtime runs into signal handler and throws exception in it. And note that currently the feature isn't supported in SGX.

Not sure what causes the performance issue, there may be several possiblities:
(1) the opt level and size level, by default they are set to 3, could you try adding --size-level=1 or 2 for wamrc?
(2) the optimization passes in aot_llvm.c, we removed several passes since from our testing, they are not important. Could you try resuming them as before:

    LLVMAddBasicAliasAnalysisPass(comp_ctx->pass_mgr);
    LLVMAddPromoteMemoryToRegisterPass(comp_ctx->pass_mgr);
    LLVMAddInstructionCombiningPass(comp_ctx->pass_mgr);
    LLVMAddJumpThreadingPass(comp_ctx->pass_mgr);
    LLVMAddConstantPropagationPass(comp_ctx->pass_mgr);
    LLVMAddReassociatePass(comp_ctx->pass_mgr);
    LLVMAddGVNPass(comp_ctx->pass_mgr);
    LLVMAddCFGSimplificationPass(comp_ctx->pass_mgr);
(3) some initializations before calling aot code in invoke_native_with_hw_bound_check(), like touching each stack page, if you keep calling aot function, the initialization might be an overhead. From my testing, it seems not an issue, the first time it costs some time, but the second time, it is much faster. Could you try the patch upload?
[hardware_trap.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/5009747/hardware_trap.zip)

ZenLulz, 2020-08-03:
Hey @wenyongh ! Thanks for your answer !

My benchmarking application embeds WAMR API to execute the AoT'ed WASM. The workflow is the following:
 
1) launching the application with the runtime embedded;
2) resolve my benchmark function using `wasm_runtime_lookup_function`;
3) get the current time using `clock()`, call the resolved function using `wasm_runtime_call_wasm` and finally get the current time using `clock()` again.

Therefore, I perform just one call to `wasm_runtime_call_wasm` per measurement. The call overhead should be absorbed in that case. Note that I'm using SQLite for the benchmarking, which puts a lot of pressure on the memory (I use the pool allocator).

Thanks for your various suggestions, give me some time to try out what you propose and I come back to you !

Cheers !

closed 2020-12-24 by wenyongh

littleblackblack, 2021-04-19:
I am facing the same problem, and I have verified that wenyongh's patch. It has some effects. The best performance is to turn off HW boundary check. It's not clear where performance is lost

wenyongh, 2021-04-20:
Hi, we found that the aot function calling procedure, or the preparation phase to call the AOT function (before actually call the AOT function pointer) is relatively slow if we frequently call the AOT function from host side. We have refined the calling procedure in patch #617 which should improve the efficiency a lot, could you please try again?