wasm-micro-runtime/wasm-micro-runtime #334: wamrc compile error on macOS

AfflatusX, 2020-08-04:
First of all, thanks for the awesome work - I was really amazed by the fact that I was able to get the VM + wasm running on my RT1050 board with very little issues!

After getting basic .wasm file running on the board, I want to give AOT a try. However, I'm stuck on the build step on macOS:

```
[ 98%] Linking CXX executable wamrc
Undefined symbols for architecture x86_64:
  "_wasm_cluster_create", referenced from:
      _wasm_exec_env_create in libvmlib.a(wasm_exec_env.c.o)
  "_wasm_cluster_del_exec_env", referenced from:
      _wasm_exec_env_destroy in libvmlib.a(wasm_exec_env.c.o)
  "_wasm_cluster_destroy_spawned_exec_env", referenced from:
      _wasm_runtime_destroy_spawned_exec_env in libvmlib.a(wasm_runtime_common.c.o)
      _wasm_runtime_spawn_thread in libvmlib.a(wasm_runtime_common.c.o)
      _wasm_runtime_thread_routine in libvmlib.a(wasm_runtime_common.c.o)
  "_wasm_cluster_set_max_thread_num", referenced from:
      _wasm_runtime_full_init in libvmlib.a(wasm_runtime_common.c.o)
      _wasm_runtime_set_max_thread_num in libvmlib.a(wasm_runtime_common.c.o)
  "_wasm_cluster_spawn_exec_env", referenced from:
      _wasm_runtime_spawn_exec_env in libvmlib.a(wasm_runtime_common.c.o)
      _wasm_runtime_spawn_thread in libvmlib.a(wasm_runtime_common.c.o)
  "_wasm_cluster_terminate_all_except_self", referenced from:
      _wasm_exec_env_destroy in libvmlib.a(wasm_exec_env.c.o)
  "_wasm_exec_env_get_cluster", referenced from:
      _wasm_exec_env_destroy in libvmlib.a(wasm_exec_env.c.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [wamrc] Error 1
make[1]: *** [CMakeFiles/wamrc.dir/all] Error 2
```

I followed the instruction of using cmake -DWAMR_BUILD_PLATFORM=darwin ..  Any idea what might be going on here? I did some search and looks like all these functions without the "_" fix exists.

xwang98, 2020-08-04:
You need to set -DWAMR_BUILD_PLATFORM=[target platform]. It can be one of subfolder names under folder:
https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/core/shared/platform. Your target board should not be running MacOS, right?

It looks you need to set the wamrc argument for your target architecture. Refer to the document here:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wasm_app.md#compile-wasm-to-aot-module

AfflatusX, 2020-08-04:
Thanks for the quick reply! I'm actually referring to the step before actually building AOT:

`
Build wamrc AoT compiler
Both wasm binary file and AoT file are supported by iwasm. The wamrc AoT compiler is to compile wasm binary file to AoT file which can also be run by iwasm. Execute following commands to build wamrc compiler:

cd wamr-compiler
./build_llvm.sh (use build_llvm_xtensa.sh instead to support xtensa target)
mkdir build && cd build
cmake ..
make
ln -s {current path}/wamrc /usr/bin/wamrc
For MacOS, you should replace cmake .. with cmake -DWAMR_BUILD_PLATFORM=darwin ...

`

I got above error when executing "make" step.

AfflatusX, 2020-08-05:
Looks like this is an issue with latest - I tried with WAMR-06-15-2020 branch and the issue went away. Regression maybe?

wenyongh, 2020-08-05:
@AfflatusX thanks for reporting the issue! It is introduced by patch #333 , we have fixed it in patch #335 , please try again.

AfflatusX, 2020-08-05:
Sure, I'll give it another shot tomorrow morning.

AfflatusX, 2020-08-05:
@wenyongh - can confirm getting latest code fixed the problem as well. I'll close the issue!

So basically I JUST missed the patch in my first try lol

closed 2020-08-05 by AfflatusX