wasm-micro-runtime/wasm-micro-runtime #337: Rust WASM reports "allocate memory failed."

AfflatusX, 2020-08-05:
I tried to build a very simple module in Rust and compiled into .wasm. However, when trying to load that through wasm-micro-runtime, it says "allocate memory failed".

I do notice that the binary from Rust is much smaller than the one from AssemblyScript. However, by reading wamr's source code, looks like the vm handles memory allocation?

Am I missing some steps here?

xujuntwt95329, 2020-08-06:
Hi,

> I do notice that the binary from Rust is much smaller than the one from AssemblyScript. However, by reading wamr's source code, looks like the vm handles memory allocation?

Yes, there are two kind of memory allocations. You can configure the memory allocation method during runtime initialization:
``` C
#if USE_GLOBAL_HEAP_BUF != 0
    init_args.mem_alloc_type = Alloc_With_Pool;
    init_args.mem_alloc_option.pool.heap_buf = global_heap_buf;
    init_args.mem_alloc_option.pool.heap_size = sizeof(global_heap_buf);
#else
    init_args.mem_alloc_type = Alloc_With_Allocator;
    init_args.mem_alloc_option.allocator.malloc_func = malloc;
    init_args.mem_alloc_option.allocator.realloc_func = realloc;
    init_args.mem_alloc_option.allocator.free_func = free;
#endif
```

Are you using `Alloc_With_Pool` mode ? Seems your memory buffer is not big enough to instantiate your wasm file, could you enlarge the `global_heap_buf` and try again?

AfflatusX, 2020-08-07:
Thanks @xujuntwt95329 ! Indeed that was the issue. Though it's weird that same code runs fine with AssemblyScript, but ran out of memory with Rust in WASM...

closed 2020-08-07 by AfflatusX