wasm-micro-runtime/wasm-micro-runtime #339: Using Zephyr example, setting WAMR_BUILD_APP_FRAMEWORK 1 results in magic header not detected.

AfflatusX, 2020-08-05:
I added  

`set (WAMR_BUILD_APP_FRAMEWORK 1)
set (WAMR_BUILD_APP_LIST WAMR_APP_BUILD_BASE)`

to the zephyr example in order to use app framework libs such as timer. However, the same AOT file now reports magic header not detected.

is app framework supported for zephyr + thumbv7?

xujuntwt95329, 2020-08-06:
Hi

Seems your runtime is built without AOT support, could you add `set (WAMR_BUILD_AOT 1)` and try again.

AfflatusX, 2020-08-06:
Hi @xujuntwt95329  thanks for the reply! That flag was on actually (otherwise I wouldn't be able to AOT my app I think?). 

Are you saying the App Framework itself was built without AOT support?

I thought that App Framework is just bunch of native API exports?

AfflatusX, 2020-08-06:
Here is my CMakeList:

`
# Copyright (C) 2019 Intel Corporation.  All rights reserved.
# SPDX-License-Identifier: Apache-2.0 WITH LLVM-exception

cmake_minimum_required(VERSION 3.8.2)

include($ENV{ZEPHYR_BASE}/cmake/app/boilerplate.cmake NO_POLICY_SCOPE)
project(NONE)

set (WAMR_BUILD_PLATFORM "zephyr")

enable_language (ASM)

add_definitions(-DWA_MALLOC=wasm_runtime_malloc)
add_definitions(-DWA_FREE=wasm_runtime_free)

# Build as THUMB by default
# change to "ARM[sub]", "THUMB[sub]", "X86_32", "MIPS" or "XTENSA"
# if we want to support arm_32, x86, mips or xtensa
if (NOT DEFINED WAMR_BUILD_TARGET)
  set (WAMR_BUILD_TARGET "THUMBV7")
endif ()

if (NOT DEFINED WAMR_BUILD_INTERP)
  # Enable Interpreter by default
  set (WAMR_BUILD_INTERP 1)
endif ()

if (NOT DEFINED WAMR_BUILD_AOT)
  set (WAMR_BUILD_AOT 1)
endif ()

if (NOT DEFINED WAMR_BUILD_JIT)
  # Disable JIT by default.
  set (WAMR_BUILD_JIT 0)
endif ()

if (NOT DEFINED WAMR_BUILD_LIBC_BUILTIN)
  # Enable libc builtin support by default
  set (WAMR_BUILD_LIBC_BUILTIN 1)
endif ()

if (NOT DEFINED WAMR_BUILD_LIBC_WASI)
  # Disable libc wasi support by default
  set (WAMR_BUILD_LIBC_WASI 0)
endif ()

if (NOT DEFINED WAMR_BUILD_APP_FRAMEWORK)
  set (WAMR_BUILD_APP_FRAMEWORK 1)
endif ()

if (NOT DEFINED WAMR_BUILD_APP_LIST)
  set (WAMR_BUILD_APP_LIST WAMR_APP_BUILD_BASE)
endif ()

set (WAMR_ROOT_DIR ${CMAKE_CURRENT_SOURCE_DIR}/wamr)

include (${WAMR_ROOT_DIR}/build-scripts/runtime_lib.cmake)

target_sources(app PRIVATE
               ${WAMR_RUNTIME_LIB_SOURCE}
               src/main.c src/native_api.c src/benchmark.c)

`

AfflatusX, 2020-08-06:
Oh also - same code without AOT works fine (with WAMR_BUILD_APP_FRAMEWORK set to 1)

xujuntwt95329, 2020-08-10:
Hi @AfflatusX 

Sorry, I think I miss understood your issue.

> I thought that App Framework is just bunch of native API exports? 

Yes, app framework is a group of native APIs, and it should have nothing to do with AoT.

> The key here is that the AOT works when WAMR_BUILD_APP_FRAMEWORK is not set to 1.

It's very strange. I can't find out the problem with your `CMakeLists`. The `magic header not detected` error should be reported by `wasm_runtime_load`, could you please add `#warning "WAMR AOT is enabled"` after [wasm_runtime_common.c:657](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/common/wasm_runtime_common.c#L657) to check if `aot_loader` is built into your binary during compilation?

closed 2020-12-10 by wenyongh