wasm-micro-runtime/wasm-micro-runtime #349: Is there a way to monitor the RAM or CPU usage of wasm app and runtime?

dpinthinker, 2020-08-13:
We need to test the cost of WAMR' app and runtime, like RAM or CPU usage. [Performance wiki](https://github.com/bytecodealliance/wasm-micro-runtime/wiki/Performance) shows some data of performance. Could you tell us the way they has been tested?

wenyongh, 2020-08-18:
For the memory usage, there is function gc_dump_heap_stats() which can be used to dump the heap profiling result, please init runtime memory allocator with global heap buffer, and ref to the patch file uploaded. The results might be like:
```bash
heap: 0x55cafbfe92f8, heap start: 0x55cafbfe946c
total free: 15984, current: 16008, highmark: 376
g_total_malloc=628408, g_total_free=309104, occupied=319304

heap: 0x55cafbf95740, heap start: 0x55cafbf958b4
total free: 10485384, current: 10485384, highmark: 471704
g_total_malloc=628408, g_total_free=629280, occupied=18446744073709550744
```
Please ignore the first three lines as they are the results of app heap, which is a region inside global heap. The `highmark` means the peak of memory usage.

For the performance, we normally measure the time usage of workload, or the score reported by workload, e.g. CoreMark.

[mem_profiling.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/5087758/mem_profiling.zip)

dpinthinker, 2020-08-18:
It seems that this method can't profile single wasm app but all the runtime.

wenyongh, 2020-08-20:
Yes, it is not easy to profile a single wasm app and WAMR hasn't supported yet. But maybe you can use the function to roughly profile the memory usage, for example, see the heap highmark changing before and after loading module, instantiating module, creating exec env, calling wasm function, and so on.

wenyongh, 2020-09-29:
@dpinthinker the memory profiling tool was recently implemented, please ref to [enable memory profiling](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wamr.md#enable-memory-profiling-experiment) for the guide, hope it is helpful for you, thanks.

dpinthinker, 2020-09-29:
@wenyongh thanks, we will try it later!

closed 2021-01-25 by wenyongh