wasm-micro-runtime/wasm-micro-runtime #351: Question about the structure definition in wgl

takoyakiyo, 2020-08-13:
I want to ask why used `typedef void lv_obj_t` instead of original definition in `lvgl`? Is it necessary to do this or can I still use the original definition？If I want to use a variable of `lv_style_t` in `wgl app`, set up its internal member variables, pass it into the function as a parameter, and call the corresponding `lvgl` function in native, then can I use the original definition of `lv_style_t` in `lvgl` instead of `void`?

xwang98, 2020-08-18:
@takoyakiyo some detailed explaination can be found here https://github.com/lvgl/lvgl/issues/1523.

takoyakiyo, 2020-08-19:
> @takoyakiyo some detailed explaination can be found here [lvgl/lvgl#1523](https://github.com/lvgl/lvgl/issues/1523).

Thanks.
I used serialization to process `style`.  And is there any way `wgl` can support `lvgl task create`?

xwang98, 2020-09-10:
Could you paste the code sample for `lvgl task create` here or give a link to the reference code? so we can have a look?

closed 2020-12-24 by wenyongh