wasm-micro-runtime/wasm-micro-runtime #352: Question about AOT sections used in App Manager

AfflatusX, 2020-08-14:
Hi everyone!

When reading Zephyr LVGL example, I noticed that to install the app, the example uses host-tool to send the AOT binary as a request_t (wrapped using init_request) to App Manager.

In target app manager, wasm_app_module_install is eventually called and try to load the payload with wasm_runtime_load_from_sections.

I tried to mimic this flow by generating an AOT version of my WASM code (which runs fine if I use wasm_runtime_load to load the file directly), wrap that using init_request, then call am_dispatch_request() to push the app to App Manager. However, that didn't work out - it reports "AOT module load failed: section missing.". 

After inspecting the payload, looks like version and magic number are there, but sections doesn't contain anything.

Looks like there is one step in wasm_runtime_load that actually creates the section_list. Do I need to manually generate the section_list when sending to the App Manager? 

Any hint/suggestions would be great!

wenyongh, 2020-08-18:
Hi, the app manager receives WASM file or AOT file which contains sections, the section link list is created by app manager when handing bytes received (ref to wasm_app_module_on_install_request_byte_arrive()), you don't need to create the section list by yourself, but just send the whole WASM file or AOT file as payload to app manager.

AfflatusX, 2020-08-18:
Ah okay - I was not sending the AOT file through wasm_app_module_on_install_request_byte_arrive, that's what was missing. Thanks for the reply!

closed 2020-08-18 by AfflatusX