wasm-micro-runtime/wasm-micro-runtime #357: Question about the memory page size of WAMR

oubotong, 2020-08-20:
I have a question about the basic page size WAMR use to allocate memory to WASM app. If WAMR follows the WebAssembly standard, the page size is 64KB and every memory allocation is a multiply of 64KB. However, this might not be good for many embedded devices since they usually have small KB-size RAM or flash. Is there anyway provided by warm to allocate memory in a smaller granularity(like 16kb page size and the memory can grow every 1 kb)?

wenyongh, 2020-08-21:
Yes, the WebAssembly standard specifies that each page size is 64KB, which might be a big memory consumption for embedded device. WAMR applies an optimization for that: if the wasm app doesn't enlarge memory (no opcode memory.grow and memory.size), and exports global __data_end and __heap_base, then runtime assumes that the linear memory space after __heap_base isn't used, and will shrink the linear memory to the size of __heap_base + app heap size. The app heap size is passed to wasm_runtime_instantiate, and the app heap is used to allocate memory for wasm app and runtime embedder to use.

Normally you can add the option **-nostdlib -Wl,--export=__heap_base,--export=__data_end -Wl,--no-entry -Wl,--allow-undefined** to generate the wasm app, e.g.:
```bash
/opt/wasi-sdk/bin/clang -O3 -nostdlib \
    -z stack-size=8192 -Wl,--initial-memory=65536 \
    -Wl,--export=main -o test.wasm test.c \
    -Wl,--export=__heap_base,--export=__data_end \
    -Wl,--no-entry -Wl,--strip-all -Wl,--allow-undefined
```
Please refer to [build wasm app](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wasm_app.md) and [embed wamr](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/embed_wamr.md) for more details. Thanks.

AfflatusX, 2020-08-25:
is there an example of how to achieve this with AssemblyScript? I didn't find --export compiler options from their official doc.

oubotong, 2020-08-25:
> The __data_end, __heap_base and stack top globals are not defined by WAMR or by wasm app source code, they are created by compiler, e.g wasi-sdk, emcc and rustc, normally the compiler divides the linear memory into three areas, and uses the wasm globals to indicates the range of each area, the layout of memory space might be:
> 
> ```shell
> -------------------------------------------------------------
> |    data area    |    stack area          |    heap area   |
> -------------------------------------------------------------
>               __data_end        __stack_top/__heap_base
>                                   <--------|---------->
>                             stack direction|heap direction
> ```
> 
> Or stack area first, data area second. The data area stores the wasm app data, e.g. C globals, strings and initial data, and __data_end global indicates the end of data area. The stack area is used in function call, the initial value of __stack_top global indicates the stack bottom, and the value of __stack_top is changed in function call. The heap area is used for dynamic memory allocation, e.g. malloc/free function, and may be dynamically enlarged if current page count isn't enough, which introduces the opcode memory.grow.
> 
> The WAMR optimization for embedded system is that if the heap area isn't used, meaning that no memory.opcode and memory.size are used in wasm app, and if __heap_base/__data_end/__stack_top global are exported in wasm app, then we might just discard the heap area, truncate the linear memory to data area + stack area (meaning the linear memory size is the value of global __heap_base). And WAMR introduces the app heap as an memory allocator, so we append the app heap to the end of linear memory, the memory space layout is:
> 
> ```shell
> -------------------------------------------------------------
> |    data area    |    stack area          |  wamr app heap  |
> -------------------------------------------------------------
>               __data_end        __stack_top/__heap_base
>                                   <--------|---------->
> ```
> 
> And we change the page size to the total memory space size but not 64KB and change the page count to 1. Since the original unused heap area's size might be large (data area and stack area might be relatively small), so after the optimization, we can reduce the total memory usage, for example: data area 4KB, stack area 8KB, app heap 8KB (data area size is set up by compiler, stack area and app heap size can be configured by developer), the total size 20KB, and before optimization, the total size must be at least 64KB.
> 
> Hope it helps, thanks.

Thanks! This clarifies my questions.

closed 2020-08-25 by oubotong