wasm-micro-runtime/wasm-micro-runtime #360: Why does wasm files loading in app manager not use wasm_runtime_load?

dpinthinker, 2020-08-25:
In modue_wasm_app.c, wasm file(bitcode or aot) is loaded by `wasm_runtime_load_from_sections`. By doing this, some sections destruction is involved.

https://github.com/bytecodealliance/wasm-micro-runtime/blob/049760b84966d119b2f95a3d9be6e17d638e2d9b/core/app-mgr/app-manager/module_wasm_app.c#L646-L659

In the project-mini, `wasm_runtime_load` is used, which seems much simpler.

dpinthinker, 2020-08-25:
And it seems that when app manager exits, all installed app will be gone.

AfflatusX, 2020-08-25:
I think the app manager code assumes the aot file was streamed from remote sources, and during the stream process, it generates sections on the fly. So by the time the app is ready to be installed, there is no reason to call wasm_runtime_load again (which generates sections again).

Personally I feel like app manager should not have assumption around how the app was received, hence it should always do wasm_runtime_load.

dpinthinker, 2020-08-25:
@AfflatusX As the issue #352 you filed, this assumption restricts app manager's extension ability.

AfflatusX, 2020-08-26:
Yeah I agree - I replaced that with wasm_runtime_load in my fork actually. However, I think it would be better to handle section generation in this routine instead of making the assumption around how the data was received :)

xujuntwt95329, 2020-08-26:
The `app-manager` is designed to be run on many different platforms. 
On some constrained devices such as MCUs, the `memory consumption`, especially the `peak memory usage` is a key factor. We use `wasm_runtime_load_from_sections` to create the module and then we can free some useless sections by `destroy_part_wasm_sections` to reduce the memory consumption and peak memory usage.

dpinthinker, 2020-08-27:
@xujuntwt95329 Could `destroy_part_wasm_sections` be import for `wasm_runtime_load` to make all platforms benefited from it? or make `wasm_runtime_load_from_sections ` and `wasm_runtime_load` same?

xujuntwt95329, 2020-08-27:
> Could destroy_part_wasm_sections be import for wasm_runtime_load to make all platforms benefited from it?

I‘m afraid not. On some platforms like `Linux`, we read the whole file into memory and call `wasm_runtime_load`, in this case, we can't destroy the section body as they don't have their own memory block.

> or make wasm_runtime_load_from_sections and wasm_runtime_load same?

Developers are free to decide which API to use. They can simply read the whole file and use `wasm_runtime_load` (just like the product-mini), or they can also resolve every section from the file, and then use `wasm_runtime_load_from_sections` to load the module, which may be a little more complicated.

dpinthinker, 2020-08-28:
`wasm_runtime_load` codes show that it also destroy loaded sections after loading sections.

https://github.com/bytecodealliance/wasm-micro-runtime/blob/034606b0a986a43f6063368e1ac28587472710bd/core/iwasm/aot/aot_loader.c#L1894-L1909

xujuntwt95329, 2020-08-28:
The `destroy_sections` only destroy the memory space of `AOTSection` or `WASMSection` structure, `section_body` is not freed.

closed 2021-02-23 by wenyongh