wasm-micro-runtime/wasm-micro-runtime #363: Getting "Exception: out of bounds memory access" when upgrading from older commit

damgaard22, 2020-08-31:
I am having problems updating my code to the newest version of WAMR.
I am running WAMR inside of SGX.
My code worked on commit cfcaca3.

I run into `Exception: out of bounds memory access`
Possibly caused by `No aux stack info for this module, can't create thread`

My code is mostly the SGX example from cfcaca3 mixed with the example on passing buffers to WASM. 

```
#include <bh_platform.h>
#include <wasm_export.h>
#include "Enclave_t.h"

extern "C" {
    typedef void (*os_print_function_t)(const char* message);
    extern void os_set_print_function(os_print_function_t pf);

    void
    enclave_print(const char *message)
    {
        ocall_print(message);
    }
}

/* TODO: Dynamically allocate with regard to data size */
static char global_heap_buf[100 * 1024 * 1024] = { 0 };

static int app_argc;
static char **app_argv;

static void
set_error_buf(char *error_buf, uint32 error_buf_size, const char *string)
{
    if (error_buf != NULL)
        snprintf(error_buf, error_buf_size, "%s", string);
}

uint8_t* execute(char *executor, char *code, size_t code_size, char *data, size_t data_size, char *enclave_uuid) {
    uint8_t *wasm_file_buf = NULL;

    /* TODO: Dynamically allocate with regard to data size */
    uint32_t stack_size = 64 * 8092, heap_size = 64 * 8092;
    wasm_function_inst_t wasm_func_execute;
    wasm_exec_env_t exec_env;
    wasm_module_t wasm_module = NULL;
    wasm_module_inst_t wasm_module_inst = NULL;
    char error_buf[128];
    char *wasm;
    size_t size_wasm;
    bool alloc_with_pool;
    uint32 max_thread_num;
    RuntimeInitArgs init_args;

    os_set_print_function(enclave_print);
    bh_log_set_verbose_level(5);

    alloc_with_pool = true;
    max_thread_num = 1;
    memset(&init_args, 0, sizeof(RuntimeInitArgs));
    init_args.max_thread_num = max_thread_num;
    init_args.mem_alloc_type = Alloc_With_Pool;
    init_args.mem_alloc_option.pool.heap_buf = global_heap_buf;
    init_args.mem_alloc_option.pool.heap_size = sizeof(global_heap_buf);	

   if (!wasm_runtime_full_init(&init_args)) {
        ocall_print("Init runtime environment failed.\n");
        return (uint8_t *) "";
    }

    /* read wasm file into bytes and store in "wasm" */
    ocall_read_file("test.wasm", (unsigned char **) &wasm, (size_t * ) & size_wasm);

    /* initialize runtime environment */
    if (!wasm_runtime_init())
        goto fail1;

    /* load WASM module */
    if (!(wasm_module = wasm_runtime_load((uint8_t *) wasm, size_wasm,
                                          error_buf, sizeof(error_buf)))) {
        ocall_print(error_buf);
        ocall_print("\n");
        goto fail2;
    }

    /* instantiate the module */
    if (!(wasm_module_inst = wasm_runtime_instantiate(wasm_module,
                                                      heap_size,
                                                      stack_size,
                                                      error_buf,
                                                      sizeof(error_buf)))) {
        ocall_print(error_buf);
        ocall_print("\n");
        goto fail3;
    }

    /* create wasm execution environment with specified stack size */
    exec_env = wasm_runtime_create_exec_env(wasm_module_inst, stack_size);

    wasm_func_execute = wasm_runtime_lookup_function(wasm_module_inst, "execute_function", NULL);

    int32_t runtime_allocated_code;
    int32_t runtime_allocated_data;

    runtime_allocated_code = wasm_runtime_module_dup_data(wasm_module_inst, code, code_size);
    runtime_allocated_data = wasm_runtime_module_dup_data(wasm_module_inst, data, data_size);

    ocall_print(std::to_string(runtime_allocated_code).c_str());
    ocall_print(std::to_string(runtime_allocated_data).c_str());

    ocall_print("Copied data into runtime");

    if (runtime_allocated_code != 0) {
        
        ocall_print("Setting args");

        uint32_t argv[5];
        argv[0] = 8; // return pointer in Wasm address space
        argv[1] = runtime_allocated_code;    // pass the buffer offset for the code in WASM space.
        argv[2] = code_size;
        argv[3] = runtime_allocated_data;  // pass the buffer offset for the input data in WASM space.
        argv[4] = data_size;

        ocall_print("Set args");
        
        wasm_runtime_call_wasm(exec_env, wasm_func_execute, 5, argv);

        ocall_print("Ran Wasm function");

        const char *wasm_runtime_exception = wasm_runtime_get_exception(wasm_module_inst);
        if (wasm_runtime_exception != NULL) {
            ocall_print("Printing wasm_exception");
            ocall_print(wasm_runtime_exception);
        }

        uint32_t wasm_return_pointer = argv[0];

        uint32_t wasm_result_pointer_address = wasm_return_pointer;
        uint32_t wasm_length_address = wasm_return_pointer + 4;

        uint32_t length = *(uint32_t *) wasm_runtime_addr_app_to_native(wasm_module_inst,
                                                                        wasm_length_address);
        uint32_t result_pointer = *(uint32_t *) wasm_runtime_addr_app_to_native(wasm_module_inst,
                                                                                wasm_result_pointer_address);
        uint8_t *result = (uint8_t *) wasm_runtime_addr_app_to_native(wasm_module_inst, result_pointer);

        ocall_print((char *) result);

        // it is runtime responsibility to release the memory,
        // unless the WASM app will free the passed pointer in its code
        wasm_runtime_module_free(wasm_module_inst, runtime_allocated_code);
        wasm_runtime_module_free(wasm_module_inst, runtime_allocated_data);

        return result;

    } else {
        ocall_print("Could not instantiate runtime buffers in wasm environment");
        ocall_print(wasm_runtime_get_exception(wasm_module_inst));
        goto fail2;
    }

    /* destroy the module instance */
    wasm_runtime_deinstantiate(wasm_module_inst);

    fail3:
    /* unload the module */
    wasm_runtime_unload(wasm_module);
    return (uint8_t *) "";
    
    fail2:
    /* destroy runtime environment */
    wasm_runtime_destroy_exec_env(exec_env);
    wasm_runtime_destroy();
    return (uint8_t *) "";

    fail1:
    return (uint8_t *) "";

}
```

My Enclave Configuration is the exact same as the new SGX example:
```
<!-- Please refer to User's Guide for the explanation of each field -->
<EnclaveConfiguration>
  <ProdID>0</ProdID>
  <ISVSVN>0</ISVSVN>
  <StackMaxSize>0x100000</StackMaxSize>
  <HeapMaxSize>0x1000000</HeapMaxSize>
  <ReservedMemMaxSize>0x400000</ReservedMemMaxSize>
  <ReservedMemExecutable>1</ReservedMemExecutable>
  <TCSNum>10</TCSNum>
  <TCSPolicy>1</TCSPolicy>
  <DisableDebug>0</DisableDebug>
  <MiscSelect>0</MiscSelect>
  <MiscMask>0xFFFFFFFF</MiscMask>
</EnclaveConfiguration>
```

It seems like much of the API i used on the old commit is still supported so the only thing i changed was the initialisation of the runtime. 

With the highest log level, the  code prints:
```
[00:00:00:000 - 0]:  
Load type section success.
 
[00:00:00:000 - 0]:  
warning: fail to link import function (wbg, __wbindgen_throw) 

[00:00:00:000 - 0]:  
Load import section success.
 
[00:00:00:000 - 0]:  
Load function section success.
 
[00:00:00:000 - 0]:  
Load table section success.
 
[00:00:00:000 - 0]:  
Load memory section success.
 
[00:00:00:000 - 0]:  
Load global section success.

[00:00:00:000 - 0]:  
Load export section success.
 
[00:00:00:000 - 0]:  
Load table segment section success.

[00:00:00:000 - 0]:  
Load code segment section success.
 
[00:00:00:000 - 0]:  
Load data segment section success.
 
[00:00:00:000 - 0]:  
Load custom section success.
 
[00:00:00:000 - 0]:  
Load custom section success.
 
[00:00:00:000 - 0]:  
Load module success.
 
[00:00:00:000 - 0]:  
Memory instantiate: 

[00:00:00:000 - 0]:  
  page bytes: 65536, init pages: 28, max pages: 16392 

[00:00:00:000 - 0]:  
  heap offset: 1310720, heap size: 523264

[00:00:00:000 - 0]:  
No aux stack info for this module, can't create thread 
 
1311280 
1311952 
Copied data into runtime 
Setting args 
Set args 
Ran Wasm function 
Printing wasm_exception 
Exception: out of bounds memory access 
```
This indicates to me that the initialisation and the duplication of data buffers into WASM went well (since it returned and printed the pointer addresses).
It seems like the error comes from: `wasm_runtime_call_wasm(exec_env, wasm_func_execute, 5, argv);`

Has there been any change to passing buffers to WASM or is there anything that i used that is no longer supported?
Furthermore can you give me some pointers on what the error message: 
`No aux stack info for this module, can't create thread` 
Might mean? Or any way that i might be able to better debug this?

If there is a more suitable communication channel for this, i will delete the issue and post it there.

All in all i have been very impressed when working with WAMR for SGX, so thanks in advance!

xujuntwt95329, 2020-09-01:
Hi, thanks for reporting this issue.

> Has there been any change to passing buffers to WASM or is there anything that i used that is no longer supported?

The APIs you used are supported, I can't figure out the root cause of this issue from the information you provided, maybe we need to get more detailed information about this. Is there any global classes in your app code? Could you provide a simple wasm module which can reproduce this problem ?

I noticed that you called both `wasm_runtime_full_init` and `wasm_runtime_init`, actually you only need to use one of them. However, this may not be the root cause of this issue.

> Furthermore can you give me some pointers on what the error message:
No aux stack info for this module, can't create thread

This is not the reason. This is just a warning to indicate that you can't execute this wasm module in multiple threads, or create thread in wasm app, because there are no aux stack information in this module. If you are interested in `WAMR pthread library`, please refer to [pthread library guide](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/pthread_library.md) for more details.

mdamgaard-biolib, 2020-09-01:
Makes sense.
I am now only running wasm_runtime_full_init, but the error still persists.
I figured out that it came from `CHECK_MEMORY_OVERFLOW(bytes)` in iwasm/interpreter/wasm_interp_classic.c:229

It checks that i do not try to allocate outside of the runtime memory it seems.
```
#define CHECK_MEMORY_OVERFLOW(bytes) do { 
    uint64 offset1 = (uint64)offset + (uint64)addr;                         
    if (offset1 + bytes <= (uint64)linear_mem_size)  {                        
      /* If offset1 is in valid range, maddr must also be in valid range,   
         no need to check it again. */                                      
      maddr = memory->memory_data + offset1;                                
    } else  {                                                            
      goto out_of_bounds; 
    }
}
```
The way i expect this to fail is if i might use up all of my memory and then try to allocate 4 bytes more, then it would fail.
Though through some debug statements i find that the addr it tries to allocate 4 bytes at is `538970715`, while the linear_mem_size is only `18219008` so it does not indicate that i have used all my memory and am stepping outside of it.

For debugging purposes, here are the values when the function fails:

```
bytes = 4
offset = 12
addr = 538970715
offset1 = 538970727
linear_mem_size = 18219008
```

Could this have something to do with the `warning: fail to link import function (wbg, __wbindgen_throw)` warning?
I might be able to mock it using the Native Symbols API.

Is there anything else that could cause WAMR to suddenly operate with an address that is so far away from what is expected?
Thanks for the quick response btw.

xujuntwt95329, 2020-09-01:
``` C
bytes = 4
offset = 12
addr = 538970715
offset1 = 538970727
linear_mem_size = 18219008
```
538970715 is not a valid address in linear memory.

In your code, a hard coded address `8` is used as return pointer:
```
uint32_t argv[5];
argv[0] = 8; // return pointer in Wasm address space
          ^
argv[1] = runtime_allocated_code;    // pass the buffer offset for the code in WASM space.
argv[2] = code_size;
argv[3] = runtime_allocated_data;  // pass the buffer offset for the input data in WASM space.
argv[4] = data_size;
```

Do you write anything to the return pointer (address 8)? The layout of linear memory in most compilers looks like:
```
-------------------------------------------------------------
|    data area    |        stack area      |    heap area   |
-------------------------------------------------------------
```
 In this case the data area may be destroyed by your app code. If the original data in address 8 is the base address of a global array, then any further reference to this array will use a wrong address.

Could you please malloc a space for `return pointer` by `wasm_runtime_module_malloc` and try again?

> Could this have something to do with the warning: fail to link import function (wbg, __wbindgen_throw) warning?
I might be able to mock it using the Native Symbols API.

No, if your app try to call an unlink function, you will get `Exception: fail to call unlinked import function`. In your case, `__wbindgen_throw` is not called during execution.

> Is there anything else that could cause WAMR to suddenly operate with an address that is so far away from what is expected?

The most common case I can imagine is described above.

Hope this help.

mdamgaard-biolib, 2020-09-01:
Okay i will try that.
Passing 8 was just to pass something arbitrary that would get overwritten with the real return address when executing the WASM function. This worked on the old commit but does seem very strange and error prone.

Should i then provide the return pointer offset (The value returned from `wasm_runtime_module_malloc`) or the native pointer given by `wasm_runtime_module_malloc`?

Is this what you are referring to?
```
   void* native_return_ptr;
   int32_t runtime_allocated_return_ptr = wasm_runtime_module_malloc(wasm_module_inst, sizeof(int64_t), &native_return_ptr);

   uint32_t argv[5];
   argv[0] = runtime_allocated_return_ptr; 
   argv[1] = runtime_allocated_code;    // pass the buffer offset for the ONNX Model in WASM space.
   argv[2] = code_size;
   argv[3] = runtime_allocated_data;  // pass the buffer offset for the input data in WASM space.
   argv[4] = data_size;
```

This still fails, but seems more correct than passing 8.

mdamgaard-biolib, 2020-09-01:
It does get pretty far into executing the WASM function.
It fails on a WASM_OP_I32_STORE opcode inside of `wasm_interp_call_func_bytecode`.
Looking at the I32_STORE opcodes it definitely sticks out.
(The last one is the one that fails)
```
[00:00:00:000 - 0]:  
I32 STORE: 18153472 
 
[00:00:00:000 - 0]:  
I32 STORE: 18153472 

[00:00:00:000 - 0]:  
I32 STORE: 1330236 

[00:00:00:000 - 0]:  
I32 STORE: 538970715 
```
I don't know if this is of any help, but i will continue debugging.

wenyongh, 2020-09-01:
@mdamgaard-biolib do you use the fast interpreter, could you try classic interpreter? You can add -DWAMR_BUILD_FAST_INTER=0 for cmake when building iwasm lib. And is there any method to reproduce the issue? Or could you please upload the wasm binary file?

mdamgaard-biolib, 2020-09-01:
I get an `unreachable` error using the classic interpreter.
I will see if i can make something reproducable, most of the parameters and wasm is proprietary.

mdamgaard-biolib, 2020-09-01:
I am not sure how to debug wasm_interp_call_func_bytecode itself, as it is just a switch of Opcodes.

What can i do to help while trying to make something reproducable?

damgaard22, 2020-09-01:
I spent some time figuring out how to make it reproducible.

I have made a repo (https://github.com/damgaard22/wamr-test).
The repo is made to look as close to the WAMR linux SGX example as possible, to hopefully make it easier for you.
It is made to work on a Linux computer with SGX installed.

The repo includes 2 branches.

`old_commit`:  Everything needed to run using wamr on the  cfcaca3 commit. (This is working and produces the expected result)

`new_commit`: What i think is everything needed to run on the newest wamr commit (This fails with the `out of bounds memory access` error, that was seen before.

To see what changed between the two commits i have made a pull request so you can see a diff
https://github.com/damgaard22/wamr-test/pull/2/files
The new_commit branch is on the left and the old_commit branch is on the right. (Admittedly it probably should be the other way around.)

I have included a reproducible step by step guide below, but it is pretty easy to figure out from the repo.
Just remember to always run the build_wamr.sh script everytime you clone or change branch.

To reproduce this:
Clone the repo recursively to get the WAMR submodule
`git clone --recursive https://github.com/damgaard22/wamr-test`

This places you on the old_commit branch
Enter the repo and build wamr with the old commit
`./build_wamr.sh`
(You possibly need to run `chmod +x build_wamr.sh`)

This should checkout the old commit and build wamr

Now simply go into SimpleTest, make and run the generated app.
`cd SimpleTest`
`make`
`./SimpleTest`

Running the app executes an ONNX (https://onnx.ai/) function and should print some JSON.

Going to the new commit requires you to checkout `new commit`
`git checkout new_commit`

Go back to the root dir
`cd ..`

and run the build script again to build wamr with newest main

`./build_wamr.sh`

Now do the same process as before to make and run the app.
This should produce the aforementioned `out of bounds memory access` error.

I am very appreciative of all your help so far!
Please don't hesitate to ask if something with the repo seems wrong or if it does not make sense to you.

xujuntwt95329, 2020-09-02:
@damgaard22
Hi, thanks for your repo, we can reproduce the issue in our environment now.

I noticed that in your wasm bytecode, the address of your data (`runtime_allocated_data`) is passed to `__rust_dealloc` function, which cause the memory overflow exception. Have you freed this buffer in your app code?

damgaard22, 2020-09-02:
Can you expand on what you mean by: "Have you freed this buffer in your app code?"
I am not sure what app code you are referring to?
The code that compiles into WASM?

xujuntwt95329, 2020-09-02:
Yes, I mean the code compiles into WASM.

damgaard22, 2020-09-02:
I noticed that we are actually using a newer version of the tensor.wasm.
I pushed this newer wasm file (tensor_new.wasm) to both branches.

How can i check if `__rust_dealloc` is still the one to cause the error?

xujuntwt95329, 2020-09-02:
It is really hard to check it without source code, there are thousands of `__rust_dealloc` in the wasm bytecode. We find this problem by stepping into the interpreter and evaluate most function calls && opcodes, we find that most `__rust_dealloc` works well, but the one which cause the memory overflow exception takes `runtime_allocated_data` as an input, which means you freed the `runtime_allocated_data` in your application code.

The same problem exists in the new wasm file, I'd suggest you to check the application source code to see if  `runtime_allocated_data` and `runtime_allocated_code` are freed in it.

xujuntwt95329, 2020-09-03:
@damgaard22 
We find another issue in `app heap` and fixed it in PR #368, could you please update `WAMR` to the newest version?

> Note: The other issue in `app heap` is not the root cause of your issue. You still need to check the source code to ensure that  `runtime_allocated_data` and `runtime_allocated_code` are not freed in your application code.

Thanks

damgaard22, 2020-09-03:
I updated build_wamr.sh to always pull the newest commit before building.

Though the issue still persists.
Rust automatically deallocates variables when they go out of scope unless you borrow them.
So i think i will try to "borrow" the input buffers and hopefully not dealloc them.

Just to be clear:
Is the problem during the deallocation or is the problem that i am accessing `runtime_allocated_data` after free?

xujuntwt95329, 2020-09-03:
> Rust automatically deallocates variables when they go out of scope unless you borrow them.
So i think i will try to "borrow" the input buffers and hopefully not dealloc them.

That sounds reasonable, maybe this is the root cause. 

> Is the problem during the deallocation or is the problem that i am accessing runtime_allocated_data after free?

It's during the deallocation.

damgaard22, 2020-09-03:
When you say that you step into the debugger.
Do you use sgx-gdb or some other debugging tool?

damgaard22, 2020-09-04:
I am going to try and rewrite the wasm to use wasi over the weekend, and see if that helps.

Do you have any example of running wasi in WAMR?
I tried using the iwasm SGX example:

`./iwasm --dir=helloworld hello_world.wasm`

Where `helloworld` is a directory at the same level as iwasm.

I get the Rust error:
`failed to find a pre-opened file descriptor through which "helloworld/helloworld.txt" could be opened`

The source code is very simple:
```
use std::io::prelude::*;
use std::fs;

fn main() {
    println!("Hello world!");
    let mut file = fs::File::create("/helloworld/helloworld.txt").unwrap();
    write!(file, "Hello world!\n").unwrap();
}
```

I am thinking this is just me formatting the --dir=<dir> option incorrectly.
I could not find any example of how to provide a directory for WASI in the docs, outside of the help text from iwasm.

Hopefully using WASI and files can fix the issue of deallocating the input buffers :)

wenyongh, 2020-09-04:
@damgaard22 Rust wasi sometimes doesn't work, see #278 , we are not sure why, it might be compiler issue.

For your workload, there might a way to hack the wasi deallocate: module_malloc larger memory with extra 8 leading unused bytes, set them to 0, and pass offset = allocated_offset + 8 to wasm function, see the attachment. It works, but it is just tricky.
[Enclave.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/5175193/Enclave.zip)

damgaard22, 2020-09-04:
That is awesome!

The small example seems to work perfectly now!
For the actual example i am trying to get working i will need to use more memory than a 16 MB stack and heap.

With the newest code i get "unreachable" when changing the stack and heap size to 32MB. Is that something to do with the new offset logic or should i try debugging it with sgx-gdb?

wenyongh, 2020-09-05:
Could you confirm that the wasm app needs 32M stack to run? It is really large, meaning that the function call chain is very long, are there recursive calls? If it really needs 32M stack, then we doubt another stack is not enough: the stack area inside linear memory, which is setup by compiler, generally we call it auxiliary stack:
```
-------------------------------------------------------------
|    data area    |        stack area      |    heap area   |
-------------------------------------------------------------
```
In your wasm app, seems that it is no larger than 1M:
(global $92  (mut i32) (i32.const 1048576))
Here the mutable global should be the stack top pointer, and it is decreased when function calling.

Not sure how to configure it in your compiler. In wasi-sdk clang, it can be configured with option "-z stack-size=n". Could you please enlarge it and try again?

damgaard22, 2020-09-05:
Yeah you seem to have a point. A large stack is not needed.

When testing, what i would classify as a low to medium sized example (pushed to the new_commit branch) i get "unreachable" as well. You can try for yourself if you want on the new_commit branch.

I want to be able to debug this but i am not really able to figure out the good breakpoint placements etc. to do this.

As you have actually solved the issue at hand i would propose that i thank you and Xu Jun for your amazing work, and i will open an issue on how to effectively debug WAMR using gdb (sgx-gdb in this case) and print statements.

Splitting it out into a new issue should hopefully help people searching for debugging of WAMR. 
Do you agree with that?

closed 2020-09-06 by damgaard22