wasm-micro-runtime/wasm-micro-runtime #364: build wamrc error： undefined reference to `LLVMAddConstantPropagationPass'

ifels, 2020-09-01:
When I try to build wamrc compiler using follow steps on Ubuntu 18.04:
```
cd wamr-compiler
./build_llvm.sh
mkdir build && cd build
cmake ..
make
```

the output of ```/build_llvm.sh```  is
```
Build llvm with 4 cores
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for dlfcn.h
-- Looking for dlfcn.h - found
-- Looking for errno.h
-- Looking for errno.h - found
-- Looking for fcntl.h
-- Looking for fcntl.h - found
-- Looking for link.h
-- Looking for link.h - found
-- Looking for malloc/malloc.h
-- Looking for malloc/malloc.h - not found
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for signal.h
-- Looking for signal.h - found
-- Looking for sys/ioctl.h
-- Looking for sys/ioctl.h - found
-- Looking for sys/mman.h
-- Looking for sys/mman.h - found
-- Looking for sys/param.h
-- Looking for sys/param.h - found
-- Looking for sys/resource.h
-- Looking for sys/resource.h - found
-- Looking for sys/stat.h
-- Looking for sys/stat.h - found
-- Looking for sys/time.h
-- Looking for sys/time.h - found
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for sysexits.h
-- Looking for sysexits.h - found
-- Looking for termios.h
-- Looking for termios.h - found
-- Looking for unistd.h
-- Looking for unistd.h - found
-- Looking for valgrind/valgrind.h
-- Looking for valgrind/valgrind.h - not found
-- Looking for fenv.h
-- Looking for fenv.h - found
-- Looking for FE_ALL_EXCEPT
-- Looking for FE_ALL_EXCEPT - found
-- Looking for FE_INEXACT
-- Looking for FE_INEXACT - found
-- Looking for mach/mach.h
-- Looking for mach/mach.h - not found
[... 233 lines cut ...]
```

when run ```make```,  the error output is :
```
/home/parallels/work/wasm/wasm-micro-runtime/core/iwasm/compilation/aot_llvm.c: In function ‘aot_create_comp_context’:
/home/parallels/work/wasm/wasm-micro-runtime/core/iwasm/compilation/aot_llvm.c:1286:5: warning: implicit declaration of function ‘LLVMAddConstantPropagationPass’; did you mean ‘LLVMAddCorrelatedValuePropagationPass’? [-Wimplicit-function-declaration]
     LLVMAddConstantPropagationPass(comp_ctx->pass_mgr);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     LLVMAddCorrelatedValuePropagationPass
[ 96%] Linking C static library libaotclib.a
[ 96%] Built target aotclib
Scanning dependencies of target wamrc
[ 98%] Building C object CMakeFiles/wamrc.dir/main.c.o
[100%] Linking CXX executable wamrc
libaotclib.a(aot_llvm.c.o): In function `aot_create_comp_context':
aot_llvm.c:(.text.aot_create_comp_context+0x88e): undefined reference to `LLVMAddConstantPropagationPass'
collect2: error: ld returned 1 exit status
CMakeFiles/wamrc.dir/build.make:238: recipe for target 'wamrc' failed
make[2]: *** [wamrc] Error 1
CMakeFiles/Makefile2:183: recipe for target 'CMakeFiles/wamrc.dir/all' failed
make[1]: *** [CMakeFiles/wamrc.dir/all] Error 2
Makefile:102: recipe for target 'all' failed
make: *** [all] Error 2

```

xujuntwt95329, 2020-09-01:
Hi, thanks for reporting this.

In `build_llvm.sh`, we use the master branch of llvm rather than release branch. The `LLVMAddConstantPropagationPass` seems not exists in the latest commit of llvm.

Could you please modify line:11 of `wamr-compiler/build_llvm.sh` to 
``` bash
git clone --depth 1 --branch release/10.x https://github.com/llvm/llvm-project.git llvm
```
then delete `core/deps/llvm` and run the `build_llvm.sh` again to download the release version and build wamrc again ?

We'll upload a PR to fix this soon.

Hope this help.

xujuntwt95329, 2020-09-01:
Fixed by PR #365

ifels, 2020-09-01:
> Hi, thanks for reporting this.
> 
> In `build_llvm.sh`, we use the master branch of llvm rather than release branch. The `LLVMAddConstantPropagationPass` seems not exists in the latest commit of llvm.
> 
> Could you please modify line:11 of `wamr-compiler/build_llvm.sh` to
> 
> ```shell
> git clone --depth 1 --branch release/10.x https://github.com/llvm/llvm-project.git llvm
> ```
> 
> then delete `core/deps/llvm` and run the `build_llvm.sh` again to download the release version and build wamrc again ?
> 
> We'll upload a PR to fix this soon.
> 
> Hope this help.

Thanks, It works now.

closed 2020-09-01 by ifels