wasm-micro-runtime/wasm-micro-runtime #366: Build error for Android by set (WAMR_BUILD_LIBC_WASI     1)

ifels, 2020-09-02:
When I try to use LIBC_WASI  to build android so, I got an error.

The CMakeLists.txt is : 
```
...
set (CMAKE_VERBOSE_MAKEFILE on)
set (CMAKE_BUILD_TYPE       Release)

set (CMAKE_TOOLCHAIN_FILE "$ENV{ANDROID_NDK_HOME}/build/cmake/android.toolchain.cmake")
set (ANDROID_NDK $ENV{ANDROID_NDK_HOME})
set (ANDROID_SDK $ENV{ANDROID_SDK_HOME})
set (ANDROID_ABI "armeabi-v7a")
set (ANDROID_LD lld)

project (iwasm)

set (WAMR_BUILD_PLATFORM      "android")
set (WAMR_BUILD_TARGET        "ARM")
set (WAMR_BUILD_TYPE          Release)
set (WAMR_BUILD_INTERP        1)
set (WAMR_BUILD_AOT           1)
set (WAMR_BUILD_LIBC_BUILTIN  1)
set (WAMR_BUILD_LIBC_WASI     1)
...

```

The error output of ```make``` is :
```
-- ANDROID_PLATFORM not set. Defaulting to minimum supported version 16.
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /opt/android/ndk-21d/toolchains/llvm/prebuilt/linux-x86_64/bin/clang - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /opt/android/ndk-21d/toolchains/llvm/prebuilt/linux-x86_64/bin/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build Configurations:
     Build as target ARM
     CMAKE_BUILD_TYPE Release
     WAMR Interpreter enabled
     WAMR AOT enabled
     WAMR JIT disabled
     Libc builtin enabled
     Libc WASI enabled
     Fast interpreter disabled
     Multiple modules disabled
-- The ASM compiler identification is Clang
-- Found assembler: /opt/android/ndk-21d/toolchains/llvm/prebuilt/linux-x86_64/bin/clang
-- Configuring done
-- Generating done
-- Build files have been written to: /home/parallels/work/wasm/wasm-micro-runtime/product-mini/platforms/android/build
Scanning dependencies of target iwasm
[  2%] Building C object CMakeFiles/iwasm.dir/home/parallels/work/wasm/wasm-micro-runtime/core/shared/platform/android/platform_init.c.o
[  5%] Building C object CMakeFiles/iwasm.dir/home/parallels/work/wasm/wasm-micro-runtime/core/shared/platform/common/posix/posix_malloc.c.o
[  8%] Building C object CMakeFiles/iwasm.dir/home/parallels/work/wasm/wasm-micro-runtime/core/shared/platform/common/posix/posix_memmap.c.o
[ 11%] Building C object CMakeFiles/iwasm.dir/home/parallels/work/wasm/wasm-micro-runtime/core/shared/platform/common/posix/posix_thread.c.o
/home/parallels/work/wasm/wasm-micro-runtime/core/shared/platform/common/posix/posix_thread.c:32:72: warning: unused parameter 'prio' [-Wunused-parameter]
                               void *arg, unsigned int stack_size, int prio)
                                                                       ^
1 warning generated.
[ 13%] Building C object CMakeFiles/iwasm.dir/home/parallels/work/wasm/wasm-micro-runtime/core/shared/platform/common/posix/posix_time.c.o
[ 16%] Building C object CMakeFiles/iwasm.dir/home/parallels/work/wasm/wasm-micro-runtime/core/shared/mem-alloc/ems/ems_alloc.c.o
[ 19%] Building C object CMakeFiles/iwasm.dir/home/parallels/work/wasm/wasm-micro-runtime/core/shared/mem-alloc/ems/ems_hmu.c.o
[ 22%] Building C object CMakeFiles/iwasm.dir/home/parallels/work/wasm/wasm-micro-runtime/core/shared/mem-alloc/ems/ems_kfc.c.o
[ 25%] Building C object CMakeFiles/iwasm.dir/home/parallels/work/wasm/wasm-micro-runtime/core/shared/mem-alloc/mem_alloc.c.o
[ 27%] Building C object CMakeFiles/iwasm.dir/home/parallels/work/wasm/wasm-micro-runtime/core/shared/utils/bh_assert.c.o
[... 164 lines cut ...]
                                               ^
/home/parallels/work/wasm/wasm-micro-runtime/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/posix.c:2047:17: warning: implicit declaration of function 'readlinkat' is invalid in C99 [-Wimplicit-function-declaration]
  ssize_t len = readlinkat(pa.fd, pa.path, bufsize == 0 ? fakebuf : buf,
                ^
/home/parallels/work/wasm/wasm-micro-runtime/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/posix.c:2141:19: error: use of undeclared identifier '__syscall_slong_t'
  out->tv_nsec = (__syscall_slong_t)(in % 1000000000);
                  ^
/home/parallels/work/wasm/wasm-micro-runtime/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/posix.c:2217:13: warning: implicit declaration of function 'futimens' is invalid in C99 [-Wimplicit-function-declaration]
  int ret = futimens(fd_number(fo), ts);
            ^
/home/parallels/work/wasm/wasm-micro-runtime/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/posix.c:2336:13: warning: implicit declaration of function 'symlinkat' is invalid in C99 [-Wimplicit-function-declaration]
  int ret = symlinkat(target, pa.fd, pa.path);
            ^
/home/parallels/work/wasm/wasm-micro-runtime/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/posix.c:2747:22: warning: unused parameter 'si_flags' [-Wunused-parameter]
    __wasi_siflags_t si_flags,
                     ^
15 warnings and 1 error generated.
CMakeFiles/iwasm.dir/build.make:328: recipe for target 'CMakeFiles/iwasm.dir/home/parallels/work/wasm/wasm-micro-runtime/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/posix.c.o' failed
make[2]: *** [CMakeFiles/iwasm.dir/home/parallels/work/wasm/wasm-micro-runtime/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/posix.c.o] Error 1
CMakeFiles/Makefile2:94: recipe for target 'CMakeFiles/iwasm.dir/all' failed
make[1]: *** [CMakeFiles/iwasm.dir/all] Error 2
Makefile:102: recipe for target 'all' failed
make: *** [all] Error 2

```

xujuntwt95329, 2020-09-02:
Currently `WASI for Android` is not well supported. We have an experimental branch [here](https://github.com/xujuntwt95329/wasm-micro-runtime/tree/android_wasi) which implements some of the WASI APIs, however, most of the file system related APIs (`read`, `write`, `fd_advise`, `linkat` and so on) are not implemented yet.

bxkai, 2020-09-29:
I add “set (ANDROID_PLATFORM android-29)” to build success，and file write ok；but must firstly call “app_instance_main”，if not it's "Capabilities insufficient" error

closed 2021-02-24 by wenyongh