wasm-micro-runtime/wasm-micro-runtime #372: Naive questions on the "Performance" wiki page

gasche, 2020-09-05:
Hi there,

I ended up on the [Performance wiki page](https://github.com/bytecodealliance/wasm-micro-runtime/wiki/Performance) from the project README. But I find the numbers very hard to understand; I am opening an issue so that they could hopefully be clarified.

A minor point first: the bottom of the page says:

> Some background of the performance comparison between WAMR and wasm3:
> 
> https://github.com/bytecodealliance/wasm-micro-runtime/issues/134

But the issue 134 does not obviously seem related (it's called "XTensa / FreeRTOS support"); I suppose that you are meaning to point to specific (off-topic) comments in the middle of the issue, starting at https://github.com/bytecodealliance/wasm-micro-runtime/issues/134#issuecomment-575466157 . Giving this more precise pointer would help -- I thought at first that the issue number was wrong by mistake.

Now to the real issue: the numbers currently looking somewhat like this:

> | workload | wamr-aot / wasm3 | wasm3 / wamr-interpreter | wamr-aot / GCC(O3) | wamr interpreters (fast/classic) |
> | -------- | ---------------- | ------------------------ | ------------------ | -------------------------------- |
> | matrix   | 22.02            | 1.23                     | 0.68               | 2.47                             |

I am confused by what the columns names of the form "foo / bar" are supposed to mean. Naively I would expect that the number given is computed as a division `time-taken-by-foo / time-taken-by-bar`. For example, if the number is 10, then foo is 10x slower than bar. This suggests that:
- wamr-aot is 22x slower than wasm3
- wasm3 is 1.2x slower than wamr-interpreter (so I suppose this suggests wamr-aot is 27x slower than wamr-interpreter, which sounds surprising)
- wamr-aot is 0.68x slower than GCC, which means 1.47x faster (sounds surprising)
- "fast" wamr interpreter is 2.47x slower than the "classic" one (does not make sense)

Clearly my assumption on the meaning of "foo / bar" is wrong. It may actually mean the exact converse: the number would be `time-of-bar / time-of-foo`, or something else. I have no idea how to read this data.

I would suggest the following improvements to the way the benchmark results are presented:
- explicitly define somewhere how the numbers are computed
- if you are indeed taking ratios, consistently keep "your" implementation at the same place (divisor or dividend). Having "wamr-aot / wasm3" and then "wasm3 / wamr-interpreter" is confusing. (When comparing two implementation of yours, maybe put the best one in this spot?)
- if you can use words instead of symbols, the data may be much easier to interpret: instead of 0.86 you could say "wamr-aot is 1.47x slower than GCC", which removes any ambiguity from saying that the "wamr-aot / GCC" number is "0.68".

xwang98, 2020-09-10:
@gasche Thank you for the great feedback and suggestion! Yes, we should make it more clearly described. Here is a quick clarification - Bigger ratio number for (A/B) in the wiki page always means A performance is better than B. The coremark workload is measured its own scoring system (bigger is better), and other workloads are measured by the total time used (less is better). To ensure the ratio numbers have a consistent meaning for A/B, we have switched the numerator and denominator for the two workload types during doing division of fractions.

closed 2021-02-24 by wenyongh

gasche, 2021-02-24:
@wenyongh have the issues with the wiki page been addressed? I just had a look at the current version, and I have the impression that the problems described in this issue remain, in particular the column names of the form "foo / bar" are still very confusing. Is there any plan to address them?

wenyongh, 2021-02-24:
@gasche yes, I think that @xwang98 had provided more details and the issue was opened a little long time ago, so I closed the issue. Per your suggestion, I modified the wiki page with more description, could you please help check whether it is easier to understand? Thanks!

gasche, 2021-02-24:
The wiki page now explicitly says that `A/B` actually means `time(B)/time(A)`. I think this is very confusing, but at least it is said explicitly, thanks. This corresponds to point (1) of my three suggestions:

> I would suggest the following improvements to the way the benchmark results are presented:
- explicitly define somewhere how the numbers are computed
- if you are indeed taking ratios, consistently keep "your" implementation at the same place (divisor or dividend). Having "wamr-aot / wasm3" and then "wasm3 / wamr-interpreter" is confusing. (When comparing two implementation of yours, maybe put the best one in this spot?)
- if you can use words instead of symbols, the data may be much easier to interpret: a tabular cell could read "wamr-aot is 1.47x slower than GCC", which removes any ambiguity from saying that the "wamr-aot / GCC" number is "0.68".

I think that it would have been useful to also consider point 2 and 3, but I suppose that you don't want to take the time to improve the wiki page because it's not a priority, and this is fine by me. At least with your new change, people can understand what the numbers mean.