wasm-micro-runtime/wasm-micro-runtime #373: Debugging WAMR

damgaard22, 2020-09-06:
Hi WAMR Team,

Solving the previous issue leads me to run into the next one.
When running the new example on the repo from issue #363 i get "Unreachable".

I would like to be able to debug this effectively but i am unable to set the right breakpoints or figure out where to place my "LOG_VERBOSE" statements.
I am running under SGX and using the fast interpreter.

I would like to know if you have any tips on using sgx-gdb with WAMR.
Any patches that prints things like function calls.
Currently i am struggling with both printing what functions (inside of WASM) and individual opcodes are being run. 

I am basically asking if you could shed some light on your own debugging process.

wenyongh, 2020-09-07:
Hi, to trace which opcode in which function throws the "unreachable" exception, firstly please switch to classic interpreter (since fast interpreter runs the pre-handled code, it is not easy to know the original opcode offset in wasm file), mark down the wasm file buf when calling wasm_loader_load, and in the handling of opcode WASM_OP_UNREACHABLE, output the opcode offset, which should be frame_ip - wasm_file_buf. Please see the patch in attachment. For example, in your latest code, the offset we dumped is:
```
##opcode offset to wasm file: 0x7511e1
```

And then we can use the wasm-objdump tool provided by wabt to dump the opcodes and their offsets of the wasm file, and see what the offset dumped belongs to. To build wabt:
```bash
git clone --recursive https://github.com/WebAssembly/wabt
cmake -C wabt gcc-release
```

To dump wasm file:
```
wasm-objdump -d tensor_new.wasm > a.log
```
From the file, we can see:
```
 7511db: 20 00                      | local.get 0
 7511dd: 36 02 08                   | i32.store 2 8
 7511e0: 00                         | unreachable
 7511e1: 0b                         | end
```
And the function is:
```
 7510ca func[5081]:
 7510cb: 02 7f                      | local[0..1] type=i32
 7510cd: 23 00                      | global.get 0 <__wbindgen_export_0>
```
Here we cannot know the function name, could you please try to export all the functions when building the wasm source code? e.g. in wasi-sdk clang, we can use -Wl,--export-all flag.

And if we knows which function causes the exception, we can add logs in wasm source code to trace which line causes the issue.

[classic_interp.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/5180706/classic_interp.zip)

wenyongh, 2020-09-07:
To debug the WAMR source code, please firstly build iwasm lib in Debug mode (cmake -DCMAKE_BUILD_TYPE=Debug), and in sgx sample Makefile, modify "SGX_DEBUG ?= 0" to "SGX_DEBUG ?= 1". Then we can use sgx-gdb to debug it, after loading the executable file, normally we can set the breakpoint at main function, and run the code line by line, and after the enclave is initialized (enclave_init() in your example), meaning that the enclave so file (including iwasm lib) is loaded, we can set new breakpoints in WAMR, e.g. wasm_interp_call_func_bytecode.

damgaard22, 2020-09-07:
Cool!

I now see that the unreachable happens inside the rust panic function.
Though this does not tell me too much.

Is there a way to print the last 10 opcodes, or maybe log every opcode to a file?

Then i would be able to see what lead to the rust panic function being called, as it could have been called from many places.

xujuntwt95329, 2020-09-07:
Hi, @damgaard22 

The solution in https://github.com/bytecodealliance/wasm-micro-runtime/issues/363#issuecomment-687162021 is just tricky, we reserve 8 leading bytes when allocating memory for your `code` and `data` to avoid `out of bounds memory access` exception, but it may introduce some other undefined behaviour as we don't know what does `rust_deallocate` actually do with this 8 bytes. 
The root cause of this problem is that you use WAMR's allocator to allocate memory space, but later this space is deallocated by the application's allocator.

We find that your wasm bytecode has exported the malloc function `__wbindgen_malloc`, so we have a better way to solve this problem by using the app's memory allocator rather than WAMR's `module_malloc`, I've submitted a [PR](https://github.com/damgaard22/wamr-test/pull/3) to your repo which fixes the `unreachable` problem and works well with your wasm file. Note that some return value is not checked in this patch, you should check them carefully in your production.

xujuntwt95329, 2020-09-07:
> Is there a way to print the last 10 opcodes, or maybe log every opcode to a file?

Yes, we have a macro to print every opcode during execution, see [wasm_interp_fast,c:1120](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/interpreter/wasm_interp_fast.c#L1120):
``` C
//#define HANDLE_OP(opcode) HANDLE_##opcode:printf(#opcode"\n");h_##opcode
```

You can uncomment and copy this line to wasm_interp_classic.c:1047, then build WAMR as classic interpreter mode, then every opcode will be printed during execution.

damgaard22, 2020-09-07:
Thanks a lot @xujuntwt95329!

This seems to produce some JSON which is a great step forward, i will proceed with checking it against expected results.
I am curious as to how it works when you only set the heap size to be 16kb?

The ONNX model (code variable) should be 9264504 bytes (9 MB)
And the data should be 1019108 (1MB)

xujuntwt95329, 2020-09-07:
The `heap_size` is the size of `WAMR app heap`. Now we are using `__wbindgen_malloc` exported from your wasm code to allocate memory from `wasm linear memory` rather than `WAMR app heap`, so 16KB is enough. The `__wbindgen_malloc` will automatically grow the memory (using `memory.grow` opcode) if there is not enough memory space on the current page, the maximum space of wasm application is defined in wasm bytecode rather than runtime.

[image]

Please refer to this image, before my patch, the space for the ONNX model and code is allocated by `module_malloc` and are located in `app heap`, the size of this area is fixed and defined by `heap_size` in your code. Now we use `__wbindgen_malloc`, the spaces are moved to `wasm linear memory`, and the size can be dynamically grown during runtime.

xujuntwt95329, 2020-09-09:
@damgaard22 
BTW, are you using TensorFlow on WebAssembly?

I'm also interested in TensorFlow and I'd like to evaluate TensorFlow on WAMR, do you have any public resources about TensorFlow on WebAssembly?

Thanks

xujuntwt95329, 2020-09-24:
Hi @damgaard22 

I can reproduce your problem on my machine, but I find that there are no name information for the functions in your wasm module `tensor_new.wasm`, which makes it very hard and maybe impossible to find out the root cause. Could you please re-build the wasm file with function symbols ? 
I don't know how to generate these information from Rustc, but your previous file `tensor.wasm` has these information, maybe you can use the same compile options.

Currently it's really hard to summarize a common way to debug such kind of problems, sometimes we may print the opcode, sometimes watch some internal data in gdb, these approaches requires a good command of wasm specifications, wasm memory layout, data structures in WAMR and even the code structure of WAMR. 

We are going to evaluate source level debugging on WAMR, and after the debugging feature is done, you will be able to debug your source code just like in gdb or lldb.

damgaard22, 2020-09-24:
My bad. 
I pushed a commit with wasm that includes name information.

It makes sense that it would require many different techniques to debug these kinds of issues.
I guess there is one debug statement that would be really cool in all circumstances. 
Are there any way to print all functions instead of all opcodes?

Thanks for the response!

xujuntwt95329, 2020-09-24:
I'm going to implement a simple `back trace` feature, it will print the call stack when the code run into exception.

I‘ll update the information here once finished.

Thanks

xujuntwt95329, 2020-09-24:
> I pushed a commit with wasm that includes name information.

I can't see the commit, the latest commit is still `New ONNX test 7f35789` 17 hours ago.

damgaard22, 2020-09-24:
Ahh the push got rejected.
Should be up now ;)

The call stack feature sounds amazing!

damgaard22, 2020-09-24:
It seems that it works in classic mode!
Classic mode seems to have fixed some of the different models that were failing.

Now we have one in particular that fails, this is maybe due to size and i will probably need help.
It is an 80mb large Inception model.
I would really like some help with this one, since i wonder if WAMR can handle this kind of size.
I get an unreachable error, but can that be due to the small stack size?
There is only 15mb allocated to WAMR stack.

I have pushed the changes to the repo.

This seems to be the only model that does not work in Classic Mode.
It is also worth noting that all the models we are testing works in the Web.

I must say that WAMR has suprised me alot and it is awesome to have something that runs in SGX!

xujuntwt95329, 2020-09-27:
> It seems that it works in classic mode!

Yes, thanks to your issue, we find a problem with the fast interpreter, and we have fixed it in PR #406 . You can run your workload in fast mode with the latest version of WAMR.

> Now we have one in particular that fails, this is maybe due to size.
It is an 80mb large Inception model.
I would really like some help with this one, since i wonder if WAMR can handle this kind of size.
I get an unreachable error, but can that be due to the small stack size?
There is only 15mb allocated to WAMR stack.

It is due to size, but not the stack size. 15MB of stack size is pretty large for most of the programs. It seems the `global_heap_buf` is not big enough so the workload failed when trying to grow memory. 
We modified the global_heap_buf from 200M to 400M and it works well, could you please enlarge the `global_heap_buf` to 400M and try again?
``` CPP
static char global_heap_buf[400 * 1024 * 1024] = { 0 };
```

xujuntwt95329, 2020-09-29:
@damgaard22 

The call stack feature is finished with PR #408.

You can checkout the latest code and build WAMR with `-DWAMR_BUILD_CUSTOM_NAME_SECTION=1`, then the call stack will be dumped when exception occurs. Please refer to [build_wamr](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wamr.md#configure-debug).

mdamgaard-biolib, 2020-09-29:
That is really cool!
I tested most of our models in fast interp mode and they all work.

I have not had time to test the call stack feature yet, but it seems exciting.
I will test it out tomorrow and close this issue if it works! :)

closed 2021-01-09 by wenyongh