wasm-micro-runtime/wasm-micro-runtime #376: Why is there a big malloc in wasm_app_module_on_install_request_byte_arrive on the last section.

dpinthinker, 2020-09-08:
We have a wasm bytecode file, which is about 11k. When loading the file, it shows that the last section malloc about 9k bytes.

Below is the codes mallocing the memory.
https://github.com/bytecodealliance/wasm-micro-runtime/blob/2135badc54eb2ea937ffd370b7718ea983e501d6/core/app-mgr/app-manager/module_wasm_app.c#L1207-L1237

But I can't understand them. There are some bits operations, and few comments about it.

Could you help to give some explanation about those codes?

wenyongh, 2020-09-08:
Hi, here it is to decode the section content size (or body size) of each section: the wasm file consists of sections, each section consists of section id (1 byte) + section content size (n bytes, u32) + section content. And the section content size is encoded using [LEB128](https://en.wikipedia.org/wiki/LEB128), with byte count no larger than (bitwidth + 6)/7, (e.g. u32, max byte count is (32+6)/7) = 5), and if a byte's highest bit is 1, there is next byte, otherwise it is the last byte.

And after decoding the section content size, app manager allocates memory with the size decoded to receive the section content. The last section with 9K bytes size you mentioned should be the code section.

dpinthinker, 2020-09-08:
Thank you @wenyongh. I didn't found those sections created by encoding are be released.  Should they be released after being loaded by wasm_runtime_load_from_sections?

wenyongh, 2020-09-09:
Some sections are released after loading, see [Sections to be released after loading](https://github.com/bytecodealliance/wasm-micro-runtime/blob/5418e09712a8ba53aac1d822d639d16f20147e9f/core/app-mgr/app-manager/module_wasm_app.c#L696), and [destroy_part_wasm_sections](https://github.com/bytecodealliance/wasm-micro-runtime/blob/5418e09712a8ba53aac1d822d639d16f20147e9f/core/app-mgr/app-manager/module_wasm_app.c#L727), and the data section is released after instantiating, see [Destroy useless sections from list after instantiate](https://github.com/bytecodealliance/wasm-micro-runtime/blob/5418e09712a8ba53aac1d822d639d16f20147e9f/core/app-mgr/app-manager/module_wasm_app.c#L759). And after that, normally there is only code section left, which stores wasm bytecodes and is used when function calling. The section type definitions can be found at [wasm.h](https://github.com/bytecodealliance/wasm-micro-runtime/blob/5418e09712a8ba53aac1d822d639d16f20147e9f/core/iwasm/interpreter/wasm.h#L43).

closed 2020-12-16 by wenyongh