wasm-micro-runtime/wasm-micro-runtime #380: LLVMAddConstantPropagationPass removed from LLVM

ZenLulz, 2020-09-09:
Hi there!

I recompiled wamrc today with a fresh git clone from LLVM and it seems WAMR AoT relies on a disappeared symbol from LLVM side: **LLVMAddConstantPropagationPass**.

Usage of this symbol in WAMR: https://github.com/bytecodealliance/wasm-micro-runtime/blob/89d2937cde4f73191c1c374fdd14e8d49004394a/core/iwasm/compilation/aot_llvm.c#L1286

Commit in LLVM related to that removal: https://github.com/llvm/llvm-project/commit/486ed885339d70cd71ee55567282a43cce28d763#diff-584e91ac5824a5700ed1c9760a529b82

As I'm not an LLVM expert, I cannot provide a fix for it.

Good luck!
Cheers

wenyongh, 2020-09-09:
Hi, the similar issue was reported by #364, and was fixed with patch #365, could you please remove core/deps/llvm folder and run build_llvm.sh again?

ZenLulz, 2020-09-09:
It works with version 10 of LLVM. I didn't think to check the closed issues, sorry!

Thanks for your support!

closed 2020-09-09 by ZenLulz