wasm-micro-runtime/wasm-micro-runtime #387: How did you measure the footprint of WAMR on Zephyr

oubotong, 2020-09-18:
Hi there, I noticed that in the wiki performance, you did the test on Zephyr RTOS. I am wondering what does the number in the table mean:

workload | wamr-aot/wasm3 | wasm3/wamr-interpreter | wamr interpreters (fast/classic)
-- | -- | -- | --
matrix | 30.17 | 1.06 | 1.65
gimli | 19.13 | 1.11 | 2.03
seqhash | 21.12 | 1.15 | 1.34

If they are the memory peak usage, could you tell me how did you do the measurement.

Thanks!

oubotong, 2020-09-18:
I think the memory measurement can be referred to issue #349 
Now I am wondering if there is a way to use `gc_dump_heap_stats()` before the load of wasm_module and after the execution of a wasm application.

xujuntwt95329, 2020-09-18:
hi,

The number in the performance table is not the memory peak usage, it's the performance ratio. `A/B = n` means `A` runs `n` times faster than `B`.

We recently introduced the WAMR memory profiling tool in PR #390. You can build WAMR with `-DWAMR_BUILD_MEMORY_PROFLING=1` and use `wasm_runtime_dump_mem_consumption` to dump the memory usage. Please refer to [build_wamr.md](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wamr.md#enable-memory-profiling-experiment).

closed 2020-09-20 by oubotong