wasm-micro-runtime/wasm-micro-runtime #393: Can I export native static function to wasm module?

oubotong, 2020-09-20:
Hi, I am wondering if it's possible to export a native static function to wasm module.

wenyongh, 2020-09-20:
Hi, please ref to [export native api](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/export_native_api.md) on how to define and register native API for wasm app and how to call it in wasm app.

oubotong, 2020-09-20:
> Hi, please ref to [export native api](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/export_native_api.md) on how to define and register native API for wasm app and how to call it in wasm app.

Hi, thanks for your answer. The document only gives an example of some non-static functions. Does wamr export the static function like `static int helloworld()`  in the same way as exporting the non-static ones?

oubotong, 2020-09-20:
Oops, I closed this issue by accident

closed 2020-09-20 by oubotong

wenyongh, 2020-09-20:
It could be static function, see [libc buildin wrappers](https://github.com/bytecodealliance/wasm-micro-runtime/blob/21850aeb0a8a30d01575d9c4d0a8307a73a7d1e3/core/iwasm/libraries/libc-builtin/libc_builtin_wrapper.c#L427), and [libc wasi wrappers](https://github.com/bytecodealliance/wasm-micro-runtime/blob/21850aeb0a8a30d01575d9c4d0a8307a73a7d1e3/core/iwasm/libraries/libc-wasi/libc_wasi_wrapper.c#L71), as far as the native function and the native symbol array are defined in the same file. Do you define the static native function and native symbol array in different files?

oubotong, 2020-09-20:
> It could be static function, see [libc buildin [...]

Yes, I just noticed that my function is not defined in the same file where the native symbol array is created. There is no more error now.