wasm-micro-runtime/wasm-micro-runtime #405: Complex native API

CanYouGiveMeAName, 2020-09-27:
I add native function refer to export_native_api.md successfully. But my function is a little complex. If i put all these function in Enclave.cpp, it will be too long. I tried make another .cpp, then include it's .h. But not used. Please help. thx

xujuntwt95329, 2020-09-28:
Hi,

I don't know what the compiler actually reported as you don't provide the error message, I guess there may be two possibilities:
1. Your functions are defined as static in the other .cpp file, so the linker can't find them.
2. You don't use `extern "C"` in your cpp file so the function name is mangling by the compiler, which cause the linking error.

If this is not the root cause, could you please provide your error message?

CanYouGiveMeAName, 2020-10-23:
> Hi,
>
> I don't know what the compiler actually reported as you don't provide the error message, I guess there may be two possibilities:
>
> 1. Your functions are defined as static in the other .cpp file, [...]

I will try. Thans very much.

closed 2020-10-23 by CanYouGiveMeAName