wasm-micro-runtime/wasm-micro-runtime #416: Need help to explain the memory consumption of module inst

oubotong, 2020-10-02:
I tried to use the API `wasm_runtime_dump_mem_consumption()` to dump out the memory consumption at runtime. And I get the following results of my program:
```
Memory consumption summary (bytes):
WASM module memory consumption, total size: 957
    module struct size: 156
    types size: 70
    imports size: 200
    funcs size: 217
    tables size: 16
    memories size: 16
    globals size: 24
    exports size: 24
    table segs size: 0
    data segs size: 36
    const strings size: 198
WASM module inst memory consumption, total size: 74236
    module inst struct size: 220
    memories size: 73772
        app heap size: 8192
    tables size: 20
    functions size: 196
    globals size: 20
    exports size: 8
Exec env memory consumption, total size: 9784
    exec env struct size: 1592
        block addr cache size: 1536
    stack size: 8192
```
I am wondering why WASM module inst would consume this much of memory compared to running the program in Zephyr natively. Can someone give an explanation?

oubotong, 2020-10-02:
Another thing that confuses me is that I tried to decrease the page size of wasm app from 65536 to 8192 in `interpreter/wasm.h`. The memory consumption changed a lot when running the same program:
```
Memory consumption summary (bytes):
WASM module memory consumption, total size: 673
    module struct size: 156
    types size: 56
    imports size: 80
    funcs size: 142
    tables size: 16
    memories size: 16
    globals size: 24
    exports size: 24
    table segs size: 0
    data segs size: 36
    const strings size: 123
WASM module inst memory consumption, total size: 16808
    module inst struct size: 220
    memories size: 16428
        app heap size: 8192
    tables size: 20
    functions size: 112
    globals size: 20
    exports size: 8
Exec env memory consumption, total size: 9784
    exec env struct size: 1592
        block addr cache size: 1536
    stack size: 8192

Total memory consumption of module, module inst and exec env: 27265
Total interpreter stack used: 304
Total aux stack used: no enough info to profile
Total app heap used: 0
```

xujuntwt95329, 2020-10-03:
Hi

As you can see, the main difference between these two dump message comes from the `memories size`, it is calculated by `linear memory size` + `app heap size` + `size of the memory instance structure`.

In your two cases, the linear memory size is different. The default page size is 64KB as specified in WebAssembly SPEC, so in the first case the `memories` occupies so much memory (64K + 8K + structure); but in case 2, you redefined the default page size to 8K, then the linear memory is decreased to 8K，so the total size consumed by `memories` is 8K(linear mem) + 8K(app heap) + structure.

> And the result would even be smaller if I change the page size to 1 which means the best way to reduce the footprint is by making page size equal to 1byte?

Actually, WAMR has a mechanism to shrink memory automatically, when building your wasm app, you can pass `-Wl,--export=__heap_base,--export=__data_end` to the compiler, then WAMR will help to calculate the suitable page size and shrink the memory automatically. The meaning of these compile options can be found [here](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wasm_app.md#build-wasm-applications).

closed 2021-02-24 by wenyongh