wasm-micro-runtime/wasm-micro-runtime #418: Question about f64 WASM ops with single precision FPU.

AfflatusX, 2020-10-08:
Hi everyone - 

I'm looking for some suggestion/ideas around following use case:

We have a board that has single precision FPU support - so we enabled hard vfp compiler flag. However, using AssemblyScript, the compiled WASM instructions has a lot of f64 ops. When we encounter those, following error will be thrown: `error: AOT module load failed: resolve symbol __aeabi_dsub failed`. Which is expected since hard vfp would not compile ARM fplib into the app.

What's the best way to work around this issue? Right now I unblocked this by replacing f64 instructions in the .wat file with f32 ones. But that would break built in lib functions which uses f64 explicitly - for example, f32.toString() no longer works as expected.

We could also avoid using built in libs involve float point altogether - but that makes devX bad and we had to create bindings in C everywhere.

Any suggestions would be greatly appreciated!

xwang98, 2020-10-08:
Please try to add -lgcc and -L {your libgcc.a path} in your gcc building script.

xujuntwt95329, 2020-10-09:
Seems the compiler-rt symbol `__aeabi_dsub` has not been registered into the runtime, could you please add `REG_SYM(__aeabi_dsub),` in the `target_sym_map` in `core/iwasm/aot/arch/aot_reloc_arm.c` (or aot_reloc_thumb.c, depends on your target) and try again:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/06b045a972e3f2ceb788298aac9c58569f246f31/core/iwasm/aot/arch/aot_reloc_arm.c#L58-L76

You may also need to register other symbols for add/mul and so on.

In addition, the ABI of the runtime and your AoT module should be the same, you can specify target-ABI when building AoT module:
```
wamrc --target-abi=xxx -o xxx.aot xxx.wasm
```

closed 2020-10-30 by wenyongh