wasm-micro-runtime/wasm-micro-runtime #421: Question About Exception Handling

takoyakiyo, 2020-10-13:
I notice that the current exception is handled is by judging the return value when the `wasm_runtime_call_wasm` function is called, obtaining the exception information explicitly and outputting it：
If an exception occurs when `on_init` is called during the app's startup, `wamr` will call `on_destroy` function of the app.
If an exception occurs during running of the app after startup, no other operation is performed.
Application resources are not released in the preceding situations. Is there a better way to handle exceptions?

wenyongh, 2020-10-23:
The wasm app thread runs into queue loop after on_init is called, waiting for queue events to handle, including RESTFUL_REQUEST/RESTFUL_RESPONSE/TIMER_EVENT_WASM and other event type registered. For the REQUEST/RESPONSE/TIMER event, app manager checks the exception, print it and then clear it if it occurs, for other event type, it doesn't check the exception. Do you mean you want to handle them? You might check exception after calling g_msg_callbacks[i](m_data, queue_msg):
```C
                for (int i = 0; i < Max_Msg_Callback; i++) {
                     if (g_msg_type[i] == message_type) {
                         g_msg_callbacks[i](m_data, queue_msg);
                         return;
                     }
                 }
```

takoyakiyo, 2020-10-26:
I mean if an exception occurs during running of the app after startup, such as "out of bounds memory access" or "stack overflow", WAMR dosen't terminate the application.

wenyongh, 2020-10-30:
Yes, but which message handling causes the exception? After calling the on_init function to start up the wasm app, the app thread runs into queue message loop to wait the message, it doesn't call wasm function again until some message is received and handled, so the exception should be thrown while handing the messages, see [app_instance_queue_callback](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/app-mgr/app-manager/module_wasm_app.c#L191) function. And some reserved messages have handled the exception, e.g., REQUEST/RESPONSE/TIMER, could you please check the left messages? e.g. in module_wasm_app.c, add exception handling after [line 317](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/app-mgr/app-manager/module_wasm_app.c#L317) and [line 359](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/app-mgr/app-manager/module_wasm_app.c#L359).

closed 2021-01-25 by wenyongh