wasm-micro-runtime/wasm-micro-runtime #424: question about Static library

CanYouGiveMeAName, 2020-10-15:
I develop base linux-sgx，and get someproblem.
1) I can not include map or unorderrd_map, but could not find. I tried in intel sgx sample, map is accessable. May this project ban some feature of C++?
2) Then i tried to include a hashmap static library. But get error: undefined reference to `__assert_fail'.
3) Then i tried to make sgx-sdk in debug mode. But get error: conflicting declaration ‘typedef uintptr_t pthread_rwlock_t’.

And another question: all sgx sample is C++ project. But real code is all C. I can not find C++ feature. This confused me very much. Is there C demo?

Please help. thx~

wenyongh, 2020-10-23:
Hi,

> 1. I can not include map or unorderrd_map, but could not find. I tried in intel sgx sample, map is accessable. May this project ban some feature of C++?

File map is under sgxsdk/include/libcxx folder, could you please add it into Enclave_Include_Paths:
```bash
Enclave_Include_Paths := ... -I$(SGX_SDK)/include/libcxx
```
> 2. Then i tried to include a hashmap static library. But get error: undefined reference to `__assert_fail'.

How to include a hashmap static library, do you mean linking it to enclave? Maybe you should also link some other libraries?
> 3. Then i tried to make sgx-sdk in debug mode. But get error: conflicting declaration ‘typedef uintptr_t pthread_rwlock_t’.

Did you install the sgx-sdk with version 2.11? Currently it supports pthread_rwlock related APIs, if yes, please comment all the contents in file: core/shared/platform/linux-sgx$ vim sgx_pthread.h/c

> And another question: all sgx sample is C++ project. But real code is all C. I can not find C++ feature. This confused me very much. Is there C demo?

We haven't trie C sample, but I guess it supports C, in Makefile, there are compilation and link flags for C:
Enclave_C_Flags, App_C_Flags

Thanks.

CanYouGiveMeAName, 2020-10-23:
> Hi,
> 
> > 1. I can not include map or unorderrd_map, but could not find. I tried in intel sgx sample, map is accessable. May this project ban some feature of C++?
> 
> File map is under sgxsdk/include/libcxx folder, could you please add it into Enclave_Include_Paths:
> 
> ```shell
> Enclave_Include_Paths := ... -I$(SGX_SDK)/include/libcxx
> ```
> 
> > 1. Then i tried to include a hashmap static library. But get error: undefined reference to `__assert_fail'.
> 
> How to include a hashmap static library, do you mean linking it to enclave? Maybe you should also link some other libraries?
> 
> > 1. Then i tried to make sgx-sdk in debug mode. But get error: conflicting declaration ‘typedef uintptr_t pthread_rwlock_t’.
> 
> Did you install the sgx-sdk with version 2.11? Currently it supports pthread_rwlock related APIs, if yes, please comment all the contents in file: core/shared/platform/linux-sgx$ vim sgx_pthread.h/c
> 
> > And another question: all sgx sample is C++ project. But real code is all C. I can not find C++ feature. This confused me very much. Is there C demo?
> 
> We haven't trie C sample, but I guess it supports C, in Makefile, there are compilation and link flags for C:
> Enclave_C_Flags, App_C_Flags
> 
> Thanks.

Thanks very much. I found the reason, assert.h that under sgx does not  define `__assert_fail`. But my assert.h unsder linux defined. And create C project successfully, but do not resolve my problem.

closed 2020-10-23 by CanYouGiveMeAName