wasm-micro-runtime/wasm-micro-runtime #432: [Interpreter Performance] The WASM execution efficiency varies greatly with compilers

chongyang0515, 2020-10-26:
I tried to port the wasm running to my project, and then use the IAR and GCC compilers. The file running efficiency of the iAR compiler was much lower than that of the GCC compiler. Is the wasm running efficiency related to the compiler?

xujuntwt95329, 2020-10-30:
Hi, the different compilers may have different optimizing strategies, options, and features.

For example, we use GCC's `Labels As Values` feature to improve the opcode dispatch, which may not be supported by `IAR`.
Also, the flags passed to GCC may not work for IAR, maybe you can try more options/flags.
Or maybe GCC's optimization is more aggressive than IAR's
......

There may be many possibilities that can influence the execution efficiency. But I think the binary built by `IAR` should not be "much slower" than `GCC`, maybe you can try more flags and settings for `IAR`

chongyang0515, 2020-10-30:
Could you tell me same flags that might be relevant and I didn't notice ，I tried a lot of iar settings, but there were no options to improve efficiency，thx

xujuntwt95329, 2020-10-30:
I'm sorry I don't have much experience on IAR. If you have tried all optimization levels and get no improvement, that might means IAR has less optimization passes than GCC, but I'm not sure on that.

closed 2021-02-24 by wenyongh