wasm-micro-runtime/wasm-micro-runtime #440: Pthread example does not work, neither with built-in libc nor with wasi

MoeMahhouk, 2020-11-09:
The provided example with the compilation commands in https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/pthread_library.md does not really work.
When compiled with built-in libc option i get this error: 

> main.c:1:10: fatal error: 'stdio.h' file not found
#include <stdio.h>
         ^~~~~~~~~
1 error generated.

However, when building with libc-wasi generates the test.wasm, but when executed in the runtime "iwasm test.wasm" it prints out this error: 

> WASM module load failed: invalid section id

It would be nice if you can help my with this issue and tell me what to do to make it work espcially with libc-wasi :)

xujuntwt95329, 2020-11-09:
hi

> When compiled with built-in libc option i get this error:
>>main.c:1:10: fatal error: 'stdio.h' file not found
#include <stdio.h>
^~~~~~~~~
1 error generated.

Have you passed `--sysroot=${WAMR_ROOT}/wamr-sdk/app/libc-builtin-sysroot` to the compiler? [stdio.h](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/wamr-sdk/app/libc-builtin-sysroot/include/stdio.h) is provided by WAMR libc-builtin sysroot. You need to change `${WAMR_ROOT}` to the actual dir of your local repo.

> However, when building with libc-wasi generates the test.wasm, but when executed in the runtime "iwasm test.wasm" it prints out this error:
>>WASM module load failed: invalid section id

Seems the `LIB_PTHREAD` is not opened when building the iwasm runtime, you need to pass `-DWAMR_BUILD_LIB_PTHREAD` to enable the pthread feature:
``` bash
cd ${WAMR_ROOT}/product-mini/platforms/linux
mkdir build && cd build
cmake .. -DWAMR_BUILD_LIB_PTHREAD=1
make
```

BTW, we have provided a sample `multi-thread` under `samples/multi-thread`, you can build both the runtime and app by CMake easily:
``` bash
cd ${WAMR_ROOT}/samples/multi-thread
mkdir build && cd build
cmake ..
make
# Run
./iwasm wasm-apps/test.wasm
```
> Note: there are some bugs in wasi-sdk 9.0+ with pthread feature, please install wasi-sdk 7.0 or 8.0 if you want to try this sample.

MoeMahhouk, 2020-11-09:
> hi
>
>
> Have you passed `--sysroot=${WAMR_ROOT}/wamr-sdk/app/libc-builtin-sysroot` to the compiler? [...]

It solved the problem, apparently i moved the local repo elsewehere in the process and forgot to change the variable. my bad :+1: 

> > However, when building with libc-wasi generates the test.wasm, but when executed in the runtime "iwasm test.wasm" it prints out this error:
> > > WASM module load failed: invalid section id
> 
> Seems the `LIB_PTHREAD` is not opened when building the iwasm runtime, you need to pass `-DWAMR_BUILD_LIB_PTHREAD` to enable the pthread feature:
> 
> ```shell
> cd ${WAMR_ROOT}/product-mini/platforms/linux
> mkdir build && cd build
> cmake .. -DWAMR_BUILD_LIB_PTHREAD=1
> make
> ```

I rebuild this again with LIB_PTHREAD enabled and now after running "iwasm test.wasm" i get a new error:
[ WASM module load failed: unknown data segment ]
Should i also use older SDK version (7 or 8) for the given example in https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/pthread_library.md ?

> BTW, we have provided a sample `multi-thread` under `samples/multi-thread`, you can build both the runtime and app by CMake easily:
> 
> ```shell
> cd ${WAMR_ROOT}/samples/multi-thread
> mkdir build && cd build
> cmake ..
> make
> # Run
> ./iwasm wasm-apps/test.wasm
> ```

yes i tried that sample as well but got this error:
[05:32:39:214 - 7BF46740]: warning: fail to link import function (env, printf)
WASM module load failed: unknown data segment
And the CMakeFiles already generate the Makefile for these binaries.
Should i rewrite them to use older versions of sdk as you mentioned in the note below or should this not happen with the provided cmake files.

> > Note: there are some bugs in wasi-sdk 9.0+ with pthread feature, please install wasi-sdk 7.0 or 8.0 if you want to try this sample.

Thank you very much for the fast reply.
Best regards

xujuntwt95329, 2020-11-10:
> I rebuild this again with LIB_PTHREAD enabled and now after running "iwasm test.wasm" i get a new error:
[ WASM module load failed: unknown data segment ]
Should i also use older SDK version (7 or 8) for the given example in https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/pthread_library.md ?

Yes, you need to use older version of wasi-sdk for pthread feature. 

> [05:32:39:214 - 7BF46740]: warning: fail to link import function (env, printf)

This should not happen, in this sample the wasm app is built as builtin-libc mode, `printf` is provided by WAMR builtin-libc, have you modified `core/iwasm/libraries/libc-builtin/libc_builtin_wrapper.c`?

> WASM module load failed: unknown data segment
And the CMakeFiles already generates the Makefile for these binaries.
Should i rewrite them to use older versions of sdk as you mentioned in the note below or should this not happen with the provided cmake files.

`WASM module load failed: unknown data segment`: this is the bug of wasi-sdk above 9.0 I mentions before, the generated wasm module will access the undefined data segment during initialization, which cause this error. To use older version of wasi-sdk, you don't need to rewrite the CMakeLists.txt, just download an older release and install to `/opt`, here is the link for wasi-sdk 8.0: [linux](https://github.com/WebAssembly/wasi-sdk/releases/download/wasi-sdk-8/wasi-sdk-8.0-linux.tar.gz) [MacOS](https://github.com/WebAssembly/wasi-sdk/releases/download/wasi-sdk-8/wasi-sdk-8.0-macos.tar.gz)

MoeMahhouk, 2020-11-10:
> Yes, you need to use older version of wasi-sdk for pthread feature.
It works with the older version of wasi-sdk :+1:  
I only noticed one difference between pthread with libc-built-in and wasi.
With wasi the outputs may collide and write upon each other. However, when compiled with libc-built-in this didn't happen and the threads didnt cause any race conditions.
Is this normal?
> 
> > [05:32:39:214 - 7BF46740]: warning: fail to link import function (env, printf)
> 
> This should not happen, in this sample the wasm app is built as builtin-libc mode, `printf` is provided by WAMR builtin-libc, have you modified `core/iwasm/libraries/libc-builtin/libc_builtin_wrapper.c`?
no i didn't do any modification to the provided samples in the git repo.
I rebuilt the sample "multi-thread" again and executed it and it only returned WASM module load failed:unknown data segment.
> `WASM module load failed: unknown data segment`: this is the bug of wasi-sdk above 9.0 I mentions before, the generated wasm module will access the undefined data segment during initialization, [...]

Main issues have been solved for me
Thank you again for your help :100:

xujuntwt95329, 2020-11-10:
> I only noticed one difference between pthread with libc-built-in and wasi.
With wasi the outputs may collide and write upon each other. However, when compiled with libc-built-in this didn't happen and the threads didnt cause any race conditions.

Both `libc-builtin` and `libc-wasi` mode depends on `pthread`, so the scheduling policy is the same between these two mode. The samples are very simple so that the possibility of race conditions is relatively low.

MoeMahhouk, 2020-11-10:
thanks a lot

closed 2020-11-10 by MoeMahhouk