wasm-micro-runtime/wasm-micro-runtime #441: SetTimeout from AS is not working.

AfflatusX, 2020-11-11:
Hi all - I created a very simple app using AssmeblyScript which calls setTimeout and setInterval and print out some test message.

For some reason, setInterval behaves properly but setTimeout does not.

I can repro in both Linux Posix and Cortex-m33 builds. Any suggestions on how should I debug further?

Thanks!

AfflatusX, 2020-11-11:
And I'm using Zephyr OS for the platform.

xujuntwt95329, 2020-11-11:
Hi,

Are you using setTimeout API provided by `assembly-script/wamr_app_lib/`?

I just test it on Linux and it works well, could you please provide your test code?

AfflatusX, 2020-11-11:
Yeah I'm using that. When you say Linux, you mean linux as target platform right? I think the issue only happens on Zephyr as the OS. It's reproducible on Linux Posix emulation as well as actually building for ARM Cortex-33 and flashing onto a physical board.

After some digging, I noticed that:

bh_post_msg2 failed to unblock bh_get_msg even though the call itself succeeded (bh_post_mst2 returns true).

if reschedule_timer is called, the bh_get_msg will be properly unblocked (which is why setInterval works but setTimeout does not)

Looks like the thread APIs are platform specific, so I think it's important to reproduce it using Zephyr.

I've created a public repo here: https://github.com/AfflatusX/wamr_repro to reproduce the issue using Linux Posix set up.

Let me know if you run into any issues! The repo uses Zephyr 2.3.0 and latest WAMR release.

AfflatusX, 2020-12-03:
@xujuntwt95329 I think I found the issue:

https://github.com/bytecodealliance/wasm-micro-runtime/blob/c8b0a1cee1de04b385bac2991e0663d9ae7bb05b/core/app-framework/base/native/timer_wrapper.c#L64

On this line, the expiry time is set to 604800000 x 1000 micro sec, but that actually overflows on a 32 bit system. In fact it becomes a negative number. 

Changing that number to something doesn't overflow fixed the issue. We should probably patch this to the older versions as well, since it causes hard-to-debug thread issues on top of the setTimeout not functioning.

xujuntwt95329, 2020-12-08:
Hi, thanks a lot for reporting this issue!

Last week, we setup up the zephyr environment and reproduce the issue by your repo. 
Now the issue has been fixed, we changed the type of `useconds` parameter for `os_cond_reltimedwait` from `int` to `uint64`, and also, we have found out and fixed some bugs in `runtime_timer.c`, now your case works well on `native_posix` environment, could you please update WAMR to latest version and try again? 

BTW, we noticed that you are using `init_request` and `am_dispatch_request` to install wasm app, this is not the correct way. The app-manager is designed to "install app while receiving", we process every arriving byte during receiving to create the section list, and after install, we can drop some unused sections to save memory. We've provided a patch based on your repo to enable this mechanism, you can have a try.
[wamr_repo.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/5658010/wamr_repo.zip)

Thanks again for finding out this problem.

AfflatusX, 2020-12-09:
re init_request / am_disptach_request: thanks for pointing it out! I'll give it a try.

closed 2021-02-24 by wenyongh