wasm-micro-runtime/wasm-micro-runtime #444: Runtime Dynamic Linking

qinkunbao, 2020-11-20:
Hi,

I was wondering if you have any plan for adding support for runtime linking `dlopen` and `dlsym` for webassembly in the future? If not, may I ask what is the main difficulty in your mind?

````
void* handle = dlopen("library.wasm", 0);
int (*fun_ptr)()= dlsym(handle, "foo");
(*fun_ptr)();
````
The corresponding web assembly code.
````
i32.const 1024
i32.const 0
call 1
i32.const 1037
call 2
call_indirect (type 0)
````

xujuntwt95329, 2020-11-24:
Hi,

The biggest problem of dlsym is: you need to call a function in another module, the function index does not exists in current module, so the `call_indirect` instruction can't find the valid function instance to call.

As a workaround, we can use some runtime wrappers to achieve this. When we get the function pointer from `dlsym`, we need to use a runtime wrapper to call this function rather than invoke the function pointer directly:
``` C
typedef int (*TestFunc)(int);

int main()
{
    int ret;
    void *handle;
    TestFunc test_func;

    handle = dlopen("app.wasm", 0);
    test_func = (TestFunc)dlsym(handle, "test_func");
    // don't use ret = (*test_func)(1);
    ret = invoke_test_func(test_func, 1);
}
```

The `invoke_test_func` is a runtime wrapper, it will call the `test_func` in the other module instance and return the value. We have a patch to test this functionality, you can have a try. 
[test-dlsym.tar.gz](https://github.com/bytecodealliance/wasm-micro-runtime/files/5587997/test-dlsym.tar.gz)

You can use the `build.sh` to compile `app.wasm` and `test.wasm`, and then apply the patch to WAMR, rebuild iwasm, and run 
``` bash
iwasm test.wasm
```

However, this is just a simple test, there are some other problems. For example, you can't pass a buffer to the function from `library.wasm` as they are in different module instance, you will need to malloc a space in `library.wasm` by using `wasm_runtime_module_malloc` and copy the content of the buffer.

JeremyJiangH, 2021-01-14:
> Hi,
> 
> The biggest problem of dlsym is: you need to call a function in another module, the function index does not exists in current module, so the `call_indirect` instruction can't find the valid function instance to call.
> 
> As a workaround, we can use some runtime wrappers to achieve this. When we get the function pointer from `dlsym`, we need to use a runtime wrapper to call this function rather than invoke the function pointer directly:
> 
> ```c
> typedef int (*TestFunc)(int);
> 
> int main()
> {
>     int ret;
>     void *handle;
>     TestFunc test_func;
> 
>     handle = dlopen("app.wasm", 0);
>     test_func = (TestFunc)dlsym(handle, "test_func");
>     // don't use ret = (*test_func)(1);
>     ret = invoke_test_func(test_func, 1);
> }
> ```
> 
> The `invoke_test_func` is a runtime wrapper, it will call the `test_func` in the other module instance and return the value. We have a patch to test this functionality, you can have a try.
> [test-dlsym.tar.gz](https://github.com/bytecodealliance/wasm-micro-runtime/files/5587997/test-dlsym.tar.gz)
> 
> You can use the `build.sh` to compile `app.wasm` and `test.wasm`, and then apply the patch to WAMR, rebuild iwasm, and run
> 
> ```shell
> iwasm test.wasm
> ```
> 
> However, this is just a simple test, there are some other problems. For example, you can't pass a buffer to the function from `library.wasm` as they are in different module instance, you will need to malloc a space in `library.wasm` by using `wasm_runtime_module_malloc` and copy the content of the buffer.

Hi, 

I use your patch and `dlsym` works. nice work.

Besides, I got a question about `build.sh`, for the option `-nostdlib`. When I build wasm file use default `build.sh` in your patch and run it, an error occured:
```
Exception: fail to call unlinked import function (env, iprintf)
```
Then I remove `-nostdlib` and build again, it works perfectly.

I just wonder how these happened, and the reason for using options `-nostdlib`.

Thanks!

xujuntwt95329, 2021-01-14:
> Exception: fail to call unlinked import function (env, iprintf)

`iprintf` is not a libc API, I guess you mean `printf` ?
If you pass `-nostdlib` option when building wasm module, then you need to build runtime with `WAMR_BUILD_LIBC_BUILTIN=1`, in this case, the `printf` will be treated as an import function, WAMR's libc-builtin library has provided wrapper for `printf`

If you remove `-nostdlib` option, then the wasm module is compiled as WASI mode, the implementation of `printf` is compiled to wasm and statically linked into your wasm module.

Please refer to [build wasm app](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wasm_app.md#build-wasm-applications) for more details.

Hope this can help you.
Thanks!

JeremyJiangH, 2021-01-14:
> `iprintf` is not a libc API, I guess you mean `printf` ?
> If you pass `-nostdlib` option when building wasm module, then you need to build runtime with `WAMR_BUILD_LIBC_BUILTIN=1`, in this case, the [...]

Sorry for not read the docs carefully, this helps a lot, thanks!

JeremyJiangH, 2021-01-15:
> Hi,
>
> The biggest problem of dlsym is: you need to call a function in another module, the function index does not exists in current module, so the `call_indirect` instruction can't find the valid [...]

Hi,

I said your patch works great on my environment, but I find I was wrong. -_-

Now 2 problems encountered:

1. in `dlsym_wrapper`, I find `wasm_runtime_lookup_function`'s return value `func` is smaller than `((AOTModuleInstance*)app_module_inst)->func_ptrs.ptr`, leads to `func_index` a negative value (overflow to a very large integer), then things start to go wrong.
2. sometimes (about 1/7) problem 1 not show up, then, when carry out `wasm_runtime_call_wasm` in `invoke_test_func_wrapper`, I got this error:
```
func name: unhandled SIGSEGV, si_addr: 0xffffffffffffffe0
Floating point exception (core dumped)
```
Sorry for cannot upload screenshots about these problems. Hope for some helps.

Thanks!

konsumer, 2024-11-05:
I also very badly need this feature. I think it would really help [null0](https://github.com/notnullgames/null0). Any progress or insights about dyn-linking in wamr?

lum1n0us, 2024-11-05:
we https://github.com/bytecodealliance/wasm-micro-runtime/tree/dev/dylink

We've kept [this experimental PR](https://github.com/bytecodealliance/wasm-micro-runtime/pull/1023) in [a branch separate from the main branch](https://github.com/bytecodealliance/wasm-micro-runtime/tree/dev/dylink). As you can see, it's been quite a while since the last update from the author.

Right now, we're focusing on [a feature for instantiation linking](https://github.com/bytecodealliance/wasm-micro-runtime/tree/dev/instantiate_linking). Following that, we'll introduce a new layer for linking. Developers will be able to contribute different linkers as needed to meet their specific requirements. Dynamic linking might be reworked as well.

closed 2025-05-08 by lum1n0us