wasm-micro-runtime/wasm-micro-runtime #447: conflicting types for ‘pthread_rwlock_t’

CanYouGiveMeAName, 2020-11-24:
in sgs 2.11

 wasm-micro-runtime-WAMR-09-29-2020/core/shared/platform/linux-sgx/sgx_pthread.h:13:19: 
sgxsdk/include/tlibc/pthread.h:61

error: conflicting types for ‘pthread_rwlock_t’
 typedef uintptr_t pthread_rwlock_t;

wenyongh, 2020-11-24:
Hi, the issue has been fixed in latest code with patch #431 , could you please ref to it, and manually modify the code of WAMR-09-29-2020? Thanks.

CanYouGiveMeAName, 2020-11-25:
> Hi, the issue has been fixed in latest code with patch #431 , could you please ref to it, and manually modify the code of WAMR-09-29-2020? Thanks.

I tried the latest version on branch main：
1) build linux-sgx
2) replace new libvmlib.a and libvmlib_untrusted.a
3) change $WAMR_ROOT to new dir

pthread_rwlock_t conflicting resolved, but get new error
CP libvmlib_untrusted.a  <=  App/third_party/lib/libvmlib_untrusted.a
/usr/local/bin/ld: App/Enclave_u.o: in function `Enclave_ocall_stat':
/opt/lq/ContractEnclave/App/Enclave_u.c:623: undefined reference to `ocall_stat'
/usr/local/bin/ld: App/Enclave_u.o: in function `Enclave_ocall_getentropy':
/opt/lq/ContractEnclave/App/Enclave_u.c:767: undefined reference to `ocall_getentropy'

CanYouGiveMeAName, 2020-11-25:
> Hi, the issue has been fixed in latest code with patch #431 , could you please ref to it, and manually modify the code of WAMR-09-29-2020? Thanks.

I tried to apply patch #431 on release WAMR-09-29-2020. Do not has error Enclave_ocall_stat.

wenyongh, 2020-11-25:
Hi, function ocall_stat and ocall_getentropy are implemented in [file.c](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/shared/platform/linux-sgx/untrusted/file.c#L133), seems that file libvmlib_untrusted.a is not the latest version, have you updated it? From your log:
```
   CP libvmlib_untrusted.a <= App/third_party/lib/libvmlib_untrusted.a
```
Seems that you copied the untrusted lib from your own folder, but not from product-mini/platforms/linux-sgx/build.

For example:
```bash
cd product-mini/platforms/linux-sgx/
mkdir build && cd build
cmake ..
make   # libvmlib.a and libvmlib_untrusted.a are generated
```
And then
```bash
cd product-mini/platforms/linux-sgx/enclave-sample
make clean && make     #ensure two libs are copied from ../build to current folder
```

CanYouGiveMeAName, 2020-11-25:
> Hi, function ocall_stat and ocall_getentropy are implemented in [file.c](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/shared/platform/linux-sgx/untrusted/file.c#L133), seems that file libvmlib_untrusted.a is not the latest version, have you updated it? From your log:
> 
> ```
>    CP libvmlib_untrusted.a <= App/third_party/lib/libvmlib_untrusted.a
> ```
> 
> Seems that you copied the untrusted lib from your own folder, but not from product-mini/platforms/linux-sgx/build.
> 
> For example:
> 
> ```shell
> cd product-mini/platforms/linux-sgx/
> mkdir build && cd build
> cmake ..
> make   # libvmlib.a and libvmlib_untrusted.a are generated
> ```
> 
> And then
> 
> ```shell
> cd product-mini/platforms/linux-sgx/enclave-sample
> make clean && make     #ensure two libs are copied from ../build to current folder
> ```

It works.
Thanks

closed 2020-11-25 by CanYouGiveMeAName