wasm-micro-runtime/wasm-micro-runtime #448: Getting error "ASSERTION FAILED: gci_is_heap_valid(heap)" when call wasm_runtime_module_malloc() multiple times.

JavanZhu, 2020-11-24:
Hello, I am following the tutorial [pass-buffer-to-wasm-function](https://github.com/bytecodealliance/wasm-micro-runtime/blob/5418e09712a8ba53aac1d822d639d16f20147e9f/doc/embed_wamr.md#pass-buffer-to-wasm-function) to call wasm function and pass some string arguments to it from native, getting following error when call wasm_runtime_module_malloc() multiple times.
```
ASSERTION FAILED: gci_is_heap_valid(heap), at file wasm-micro-runtime/core/shared/mem-alloc/ems/ems_alloc.c, line 342
```

In ems_alloc.c, it looks like the values of variable `heap->heap_id` and `heap` is not equivalent.  
```
/**
 * Do some checking to see if given pointer is a possible valid heap
 * @return GC_TRUE if all checking passed, GC_FALSE otherwise
 */
int
gci_is_heap_valid(gc_heap_t *heap)
{
    if (!heap)
        return GC_FALSE;
    if (heap->heap_id != (gc_handle_t) heap)
        return GC_FALSE;

    return GC_TRUE;
}
```

So my question is:
1. How to solve it or Possible reasons for this error? 
2. The code location where the variable `heap->heap_id`  is modified

please tell me something I'm missed, Thank you very much.

wenyongh, 2020-11-25:
Hi, it seems that the heap structure is corrupted, not sure whether it is overwritten by your wasm app, or other codes e.g. wasm vm or your host embedder. What is the language of wasm source code, and what is the wasm compiler? Is there a simple way to reproduce the error?

And could you please ensure that the buffer applied from wasm_runtime_module_malloc is not overwritten?
And also the native address of buffer applied might be changed after memory.grow opcode is executed, so please **do not** remember the native address for further use, if you want to do that, you can remember the app offset, and convert it to native address every time we use it.

If you want to debug, maybe you can watch the heap->heap_id in gdb to see when it was changed. And also you can enable the heap verify feature by changing BH_ENABLE_GC_VERIFY to 1 in file core/config.h to see whether the heap data is overwritten.

JavanZhu, 2020-11-26:
Hi @wenyongh, thank you for your reply.

My ENVIRONMENT:
- platform: Darwin
- the latest commit of wasm-micro-runtime in my native machine: 
[image]
- wasm source code file + iwasm + wasm bytecode file
[invalid_heap_res.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/5600091/invalid_heap_res.zip)

I reproduce the error through following step:

1. my wasm source code (AssemblyScript), note that I pass some big strings to native runtime.

```TypeScript
import * as console from "../wamr_app_lib/console";
import * as utils from "../wamr_app_lib/utils";

export declare type addr_t = i32; 

export function strToAB(str: string): ArrayBuffer {
  return String.UTF8.encode(str, true);
}

@external("env", "dispatch_native_call")
declare function dispatch_native_call(
  funcName: ArrayBuffer,
  params: ArrayBuffer
): void;

// main function of wasm app
export function onLoad(): void {
  let funcName = strToAB(`dispatch_native_call`);
  let big_param = strToAB(big_string);
  // call native method 
  dispatch_native_call(funcName, big_param);

  **// call native method again, it will throw error "ASSERTION FAILED: gci_is_heap_valid(heap)"
  let big_param2 = strToAB(big_string2);
  dispatch_native_call(funcName, big_param2);**

}

export function dispatch_native_event(
  buffer_offset: addr_t,
  size: i32
): void {
  console.log(
    "dispatch_native_event() called: buffer_offset = [" +
      buffer_offset.toString() +
      "], size = [" +
      size.toString() +
      "]"
  );

  var buffer = new ArrayBuffer(size);
  var dataview = new DataView(buffer);

  for (let i = 0; i < size; i++) {
    dataview.setUint8(i, load<i8>(buffer_offset + i, 0, 1));
  }

  var msg = String.UTF8.decode(buffer);
  console.log("dispatch_native_event: result = [" + msg + "]");
}

let big_string = `English literature, the body of written works produced in the English language by inhabitants of the British Isles (including Ireland) from the 7th century to the present day. The major literatures written in English outside the British Isles are treated separately under American literature, Australian literature, Canadian literature, and New Zealand literature.

George Gordon Byron, 6th Baron Byron. Lord Byron English poet (1788-1824) was a leading figure in the Romantic movement.
BRITANNICA QUIZ
10 British Poets: A Quiz
[... 205 lines cut ...]
    uint32 wasm_file_size;
    uint32 stack_size = 20 * 1024 * 1024, heap_size = 20 * 1024 * 1024;
    wasm_module_t wasm_module = NULL;
   
    ....

    /* initialize runtime environment */
    if (!wasm_runtime_full_init(&init_args)) {
        printf("Init runtime environment failed.\n");
        return -1;
    }

    register_native_cb();

    bh_log_set_verbose_level(log_verbose_level);

    ...
}

```

3. when run following command, it will crash and throw error "ASSERTION FAILED: gci_is_heap_valid(heap), at file **wasm-micro-runtime/core/shared/mem-alloc/ems/ems_alloc.c, line 337"
```
./iwasm -f onLoad ../invalid_heap_res/debug_big_str.wasm
```

JavanZhu, 2020-11-26:
@wenyongh After i updated wasm-micro-runtime source code to the latest commit:
[image]
and execute the debug_big_str.wasm again, I got error
```
Exception: out of memory
```

Because it run in following code. 
[image]

I hope this information is helpful, please tell me if you need more information about this problem, thank you.

wenyongh, 2020-11-26:
@JavanZhu yes, in latest version, we move the app heap structure (head part) out of linear memory, so only app heap data pool is inside the linear memory, and also we add many checks for the heap data nodes to enhance the security. If we find that the heap data pool is corrupted, we just report error and do nothing in the app heap memory allocation/free.

The app heap is inserted or appended into the linear memory, and the __heap_base global exported or initial page count of wasm app is amended, for most compilers, this works. For your application, we found that the wasm bytecode stores data to linear memory from low address to high address, like that it initializes the whole linear memory except the data area, so it makes app heap corrupted.

And we found that it exports the memory allocation/free functions: __alloc, __retain and __release. I have a test, by using the functions instead of alloc/free memory from/to app heap, you wasm app works well. I updated the patch, could you please have a try? This is a temporary solution, we might change the behavior of module_malloc/module_free in the future: if __alloc/__retain/__release functions are found, we don't create the app heap any more, we can just call these functions to alloc/free memory from/to linear memory.

Note that you need to run iwasm with --heap-size=0 to disable the app heap:
```bash
iwasm --heap-size=0 -f onLoad debug_big_str.wasm
```

[wamr_as.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/5601602/wamr_as.zip)

JavanZhu, 2020-11-27:
Hi @wenyongh, thank you for your help and detail explanation, thank you very much. 
I have tested it, and it works well if we pass string like "response from native", but if you replace it with "{response from native"( just add a symbol "**{**" to it),  
it throws "wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_classic.c, line 3381, meet an exception Exception: unreachable" error.

```c
// native method call by AssemblyScript
void
dispatch_native_call_wrapper(wasm_exec_env_t exec_env,
                             char *func_name,
                             char *func_param)
{
    printf("dispatch_native_call_wrapper() called with: func_name = [%s], "
           "param = [%s]\n",
           func_name, func_param);

 //    call_wasm_func(get_module_inst(exec_env), exec_env, "{response from native",
//                   sizeof("response from native"));
      // following code will throw "Exception: unreachable" error
    call_wasm_func(get_module_inst(exec_env), exec_env, "{\"resCode\":0,\"resMsg\":\"Success\",\"data\":{\"varType1\":\"var content\",\"varType2\":\"var content\"}}",
                   sizeof("{\"resCode\":0,\"resMsg\":\"Success\",\"data\":{\"varType1\":\"var content\",\"varType2\":\"var content\"}}"));

}
```

wenyongh, 2020-11-28:
@JavanZhu welcome. I checked the bytecode of wasm app again and found that I wrongly understood the function argument of __retain and __release function. When calling these two functions, we can just pass the offset returned from __alloc, and no need to plus 16, so please change
```C
argv[0] = offset + 16;
to
argv[0] = offset;
```
in two places. I tested again, it works well.

JavanZhu, 2020-11-28:
> @JavanZhu welcome. I checked the bytecode of wasm app again and found that I wrongly understood the function argument of __retain and __release function. When calling these two functions, we can [...]

@wenyongh yes, it works, thank you very much.

By the way, when i updated AssemblyScript compiler of my native environment from 0.14.12 to Version 0.17.3, I noticed that export functions is changed to following functions

```
 (export "memory" (memory $0))
 (export "__new" (func $~lib/rt/pure/__new))
 (export "__renew" (func $~lib/rt/pure/__renew))
 (export "__retain" (func $~lib/rt/pure/__retain))
 (export "__release" (func $~lib/rt/pure/__release))
```

just change "__alloc" to "__new", and everything work very well. Thank you for your help.

JavanZhu, 2020-11-28:
> @JavanZhu yes, in latest version, we move the app heap structure (header part) out of linear memory, so only app heap data pool is inside the linear memory, and also we add many checks for the heap data nodes to enhance the security. If we find that the heap data pool is corrupted, we just report error and do nothing in the app heap memory allocation/free.
> 
> The app heap is inserted or appended into the linear memory, and the __heap_base global exported or initial page count of wasm app is amended, for most compilers, this works. For your application, we found that the wasm bytecode stores data to linear memory from low address to high address, like that it initializes the whole linear memory except the data area, so it makes app heap corrupted.
> 
> And we found that it exports the memory allocation/free functions: __alloc, __retain and __release. I have a test, by using the functions instead of alloc/free memory from/to app heap, you wasm app works well. I updated the patch, could you please have a try? This is a temporary solution, we might change the behavior of module_malloc/module_free in the future: if __alloc/__retain/__release functions are found, we don't create the app heap any more, we can just call these functions to alloc/free memory from/to linear memory.
> 
> Note that you need to run iwasm with --heap-size=0 to disable the app heap:
> 
> ```shell
> iwasm --heap-size=0 -f onLoad debug_big_str.wasm
> ```
> 
> [wamr_as.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/5601646/wamr_as.zip)

@wenyongh I notice that you say "Note that you need to run iwasm with --heap-size=0 to disable the app heap:", it means that I should set the parameter heap_size of wasm_runtime_instantiate() to 0. but it works if I am not set heap_size to 0. So could you please give me more detail information about the reason why set heap_size to 0?  thank you.

I will close this issue later.

```
    /* instantiate the module */
    if (!(wasm_module_inst = wasm_runtime_instantiate(wasm_module,
                                                      stack_size,
                                                      heap_size,
                                                      error_buf,
                                                      sizeof(error_buf)))) {

```

wenyongh, 2020-11-28:
Hi, from the test, if app heap is enabled, it will be corrupted by your wasm application, and wasm_runtime_module_malloc/module_free will report error if runtime detects the corruption, and not sure whether this causes unexpected behavior for your application. Currently some wasi related environment variables are allocated from app heap, so module_malloc and module_free are called if app heap is enabled. We will try to fix it in the future, allocate these variables from global heap (or system memory) but not app heap, so you don't need to disable app heap again since it isn't actually used.

closed 2020-11-28 by JavanZhu